espressif/qemu #2: Running examples/protocols/http_server/simple with wiki defaults does not respond to requests

antfobe, 2019-11-19:
Hi all.
I am thankful that with [86ba82e](https://github.com/espressif/qemu/commit/86ba82e90fff3c814c4ab4db0bd28e34cd51c759) I was able to run the hello-world example, however I'm still not able to make use of the ethernet examples, even though I'm following the [wiki](https://github.com/espressif/qemu/wiki)

Here is my qemu output: 
[output.txt](https://github.com/espressif/qemu/files/3861489/output.txt)

My sdkconfig: 
[sdkconfig.txt](https://github.com/espressif/qemu/files/3861448/sdkconfig.txt)

My commands:
`idf.py menuconfig` (manually edited latter, removed WiFi, which also threw a panic)
`idf.py build`
`./makeflash-img.sh simple simple.bin`
`/opt/esp-idf/qemu/xtensa-softmmu/qemu-system-xtensa -nographic -machine esp32 -drive file=simple.bin,if=mtd,format=raw -nic user,model=open_eth`

[image]

igrr, 2019-11-19:
Looking at the command line, you are using the user-mode networking (`-nic user`). User-mode networking in QEMU doesn't support IGMP protocol, so `ping` will not work.
And if the server is running on the ESP32, you need to explicitly tell QEMU that you would like to forward the port (`hostfwd=tcp...`). Please check QEMU documentation (https://qemu.weilnetz.de/doc/qemu-doc.html#Network-options) and the wiki (https://wiki.qemu.org/Documentation/Networking#User_Networking_.28SLIRP.29).

antfobe, 2019-11-19:
I just remembered what you said above about user-mode and ICMP, however the server is still not responding to get requests. Does it work on your end (on qemu)?

Also I fixed the error display by qemu on `hw/net/opencores_eth.c`, by changing line 61 `MII_REG_MAX` value to 17. (Probably Max overlooked this).
`
  59 /* PHY MII registers */
  60 enum {
  61     MII_REG_MAX = 17,
  62 };
`

igrr, 2019-11-19:
Yes, it works on my with end with examples/protocols/http_server/simple and `-nic user,model=open_eth,hostfwd=tcp::8082-:80` option to QEMU, then `curl http://localhost:8082`.

antfobe, 2019-11-19:
Thanks, was missing the `hostfwd=tcp::8082-:80`, I thought It would be directly accessible through the qemu machine.

closed 2019-11-19 by antfobe