espressif/qemu #5: Support for SD Card

thomasesr, 2019-12-17:
Is it possible to connect SD Card to the emulated machine?

igrr, 2019-12-17:
There are two ways to implement this:

1. QEMU includes an SD device framework, and a few implementations (see files under hw/sd/). An emulation of ESP32 SD Host could be written to plug into this framework. Without emulating all the error conditions which the real peripheral can produce, this may even be a feasible thing to undertake.

2. One of the existing implementations of the SD device is `ssi-sd.c`, which is basically "SD over SPI". This could theoretically work with the existing emulation of ESP32 SPI controller. However most likely the SPI controller emulation is missing some things to properly work with the SDSPI driver of IDF, plus some GPIO emulation would be needed for the CS line. This option seems a bit easier to get to work, but will likely perform *much much slower* than option 1 in practice due to the byte-based nature of SSI framework.

luanmm, 2020-07-08:
Hello, @igrr!

First of all, thanks for your effort in making all this projects grow up (ESP-IDF, QEMU, ...). The emulator initiative can help a lot in various scenarios of the development workflow (specially in this moment, when we have a lot of "physical limits" to work directly with the hardware and to exchange physical device versions between the team, as we are in home most of the time).

I'm making some progress using the emulator to run parts of a medium complexity project. One of its parts, however, is more tricky than the others: make the embedded database (SQLite) be functional somehow.

In the device, it runs over the SD Card protocols (MMC or SPI), but, as it is not implemented, I've used both SPIFFS or FAT FS. The problem is that, as your previewed, it is "much much slower", being unfeasible that way (at least without emulator tweaks).

There is any possibility to use SD Card or another volume mouting strategy without touching the QEMU code? It would be great, because my knowledgement about QEMU is only as user, not as technical as it has to be to help implementing it somehow.

igrr, 2020-07-15:
Hi @luanmm, if you'd like to avoid implementing new peripherals in QEMU, the one option I can suggest is to use some existing QEMU block device driver. Having poked a bit around QEMU code, i think `virtio-blk` is a good candidate. The idea is that we add `virtio` device somewhere into the ESP32 memory map, and then write a driver (to run as part of the simulated application) which will talk to this `virtio` device.

Overall, you will need to:
* implement a `virtio-blk` driver in the (simulated) application. I found a nice article which walks through the implementation of virtio-blk: https://brennan.io/2020/03/22/sos-block-device/. The final source code is linked at the end of the article. Of course, the code will not compile in ESP-IDF as is, but it is not very hard to get it to work.
* direct FAT filesystem to use this driver (through "diskio" layer in case of FAT, e.g. https://github.com/espressif/esp-idf/blob/master/components/fatfs/diskio/diskio_sdmmc.c)
* in QEMU, add virtio-blk device somewhere into the ESP32 memory map. This is similar to what we do with `openeth` Ethernet MAC, which doesn't exist in a real ESP32, but exists in the QEMU model (https://github.com/espressif/qemu/blob/0ff3da8d3c797dcf33a45c419204f39f684376cf/hw/xtensa/esp32.c#L588).

Another option is to leverage semihosting support (https://github.com/espressif/esp-idf/tree/master/examples/storage/semihost_vfs). The only issue is that ESP-IDF uses semihosting calls compatible with OpenOCD, which are based on ARM semihosting standard. QEMU for Xtensa uses a different set of semihosting calls, compatible with Xtensa Instruction Set Simulator. So some adaptation will be required on ESP-IDF side, to make semihosting VFS compatible with QEMU. One benefit of this approach, is that you won't need to use a filesystem (FAT or SPIFFS) in the application this way, which will also speed up emulation.

Unfortunately we don't have bandwidth right now to help you with any of these approaches, but please do let us know if you succeed in implementing them, or if you meet some issue.

luanmm, 2020-07-15:
Thank you very much for the tips, @igrr!

I will try to go on with one of these strategies and will share the updates on this thread as soon as I have any results.

luanmm, 2020-11-25:
Just as an update: I couldn't make it work without digging more deeply, which is something that I can't do at this time, for lack of time and specific knowledge. Thanks, again, for all the helping tips. Hope somebody can explore this topic in the future to bring a solution.