espressif/qemu #8: question: Qemu Semihosting exit from application

tucanae47, 2020-05-30:
Hello im trying to interrupt qemu execution after application finishes, im not able to find how to do this properly, so my qemu start line looks like 
`xtensa-softmmu/qemu-system-xtensa -nographic -semihosting -machine esp32 -drive file=flash_image.bin,if=mtd,format=raw`

in arm cortex m3 im able to do this with semihosting `svc` options
```
static inline void _exit_qemu() {
  register u32_t r0 __asm__("r0");
  r0 = 0x18;
  register u32_t r1 __asm__("r1");
  r1 = 0x20026;
  __asm__ volatile("bkpt #0xAB");
}
```
 i found something inside openocd for esp 
[ADP_Stopped_ApplicationExit 0x20026](https://github.com/espressif/openocd-esp32/blob/1d00e3e6984868f008f3ecc2550454e67297fb86/src/target/semihosting_common.c#L298) which seems to be the answer but im not sure how to use it. 

And i also tried `exit(0)` and `exit(1)` but qemu keeps esp application rebooting, any help and hints are appreciated

igrr, 2020-05-30:
Please try calling esp_restart in the app, and pass -no-reboot flag to QEMU.

tucanae47, 2020-06-02:
Thanks for the fast reply! I did try `-no-reboot` with `exit` and it works!

closed 2020-06-02 by igrr