espressif/qemu #10: timer ISR is getting not trigged, SPIFFS not working, Wifi?

peterus, 2020-09-24:
I am trying to start with qemu on an existing project which has already its size. Currently I commented a lot of stuff out to test step by step. I am using the ESP32 Arduino core with PlatformIO (if I have to change some settings, please let me know!).

Currently I am facing different issues:

1. I am programming a timer which should trigger every second, this is not happening. I am using this code:

```
void IRAM_ATTR onTimer()
{
	portENTER_CRITICAL_ISR(&timerMux);
	timerTick = true;
	portEXIT_CRITICAL_ISR(&timerMux);
}

void setup_timer()
{
	timer = timerBegin(0, 80, true);
	timerAlarmWrite(timer, 1000000, true);
	timerAttachInterrupt(timer, &onTimer, true);
	timerAlarmEnable(timer);
}
```

As i was writing: on the HW it is working, any Ideas how to debug this?

2. SPIFFS not working

I am using this code:
`if(!SPIFFS.begin(true))
	{`

And getting this error message:

> E (6389) SPIFFS: mount failed, -10025
> E (80510) SPIFFS: mount failed, -10025

So it is trying to format something, but it is not working out.

3. To have something similar than Wifi I think i have to use the Ethernet approach. Is this correct?

igrr, 2020-09-24:
Hi @peterus,
regarding the timer, could you try reading the timer counter to see if it changes? If it isn't changing, that means that the timer hasn't been enabled/started successfully. If it is changing but you aren't getting interrupts, it means that the issue might be related to the interrupts.

For SPIFFS, have you changed the flash mode to DOUT in your application? https://github.com/espressif/qemu/wiki#build-test-app

Regarding Wi-Fi, you are right, it is not emulated in QEMU. You may use "OpenEth" Ethernet MAC driver provided in ESP-IDF:
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/components/esp_eth/include/esp_eth_mac.h#L379
CONFIG_ETH_USE_OPENETH needs to be enabled for this driver to be available.

peterus, 2020-09-24:
Hi @igrr,

regarding the timer issue: the counter is changing, but it looks like the hw is not triggering an interrupt as the alarm value is staying the same:

> timerRead: 579458
> timerAlarmRead: 1000000
> timerRead: 39363
> timerAlarmRead: 1000000
> timerRead: 441663
> timerAlarmRead: 1000000
> timerRead: 1073767
> timerAlarmRead: 1000000
> timerRead: 1440997
> timerAlarmRead: 1000000
> timerRead: 1794301
> timerAlarmRead: 1000000

regarding the other topics: I am currently using the arduino core which does not have this setting to change the flash mode or network mode. Do you have other suggestions which i could change (localy in the core) to get this working? - changing currently to ESP-IDF is not an option for me as I am using a lot of arduino libraries which are not ported so easily.

peterus, 2020-09-25:
i decided to give ESP-IDF a try, but unfortunately SPIFFS is also not working with the settings changed:

> Adding SPI flash device
> ==69910==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:4
> load:0x3fff0034,len:7080
> load:0x40078000,len:13072
> ho 0 tail 12 room 4
> load:0x40080400,len:3896
> entry 0x40080688
> I (1126) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
> I (1165) boot: compile time 14:10:27
> ets Jun  8 2016 00:22:57
> 
> rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:4
> load:0x3fff0034,len:7080
> load:0x40078000,len:13072
> ho 0 tail 12 room 4
> load:0x40080400,len:3896
> entry 0x40080688
> I (3232) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
> I (3239) boot: compile time 14:10:27
> I (3273) boot: chip revision: 0
> I (3307) boot.esp32: SPI Speed      : 40MHz
> I (3316) boot.esp32: SPI Mode       : DOUT
> I (3318) boot.esp32: SPI Flash Size : 4MB
> W (3350) boot.esp32: PRO CPU has been reset by WDT.
> W (3354) boot.esp32: APP CPU has been reset by WDT.
> W (3363) boot.esp32: WDT reset info: PRO CPU PC=0x0
> W (3367) boot.esp32: WDT reset info: APP CPU PC=0x0
> I (3391) boot: Enabling RNG early entropy source...
> I (3461) boot: Partition Table:
> I (3468) boot: ## Label            Usage          Type ST Offset   Length
> I (3479) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (3484) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (3495) boot:  2 factory          factory app      00 00 00010000 00100000
> I (3499) boot:  3 storage          Unknown data     01 82 00110000 000f0000
> I (3526) boot: End of partition table
> I (3575) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05df4 ( 24052) map
> I (3707) esp_image: segment 1: paddr=0x00015e1c vaddr=0x3ffb0000 size=0x0217c (  8572) load
> I (3787) esp_image: segment 2: paddr=0x00017fa0 vaddr=0x40080000 size=0x00404 (  1028) load
> I (3869) esp_image: segment 3: paddr=0x000183ac vaddr=0x40080404 size=0x07c6c ( 31852) load
> I (3993) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x185b0 ( 99760) map
> I (4221) esp_image: segment 5: paddr=0x000385d8 vaddr=0x40088070 size=0x01fc8 (  8136) load
> I (4396) boot: Loaded app from partition at offset 0x10000
> I (4407) boot: Disabling RNG early entropy source...
> I (4454) cpu_start: Pro cpu up.
> I (4464) cpu_start: Application information:
> I (4467) cpu_start: Project name:     LoRa_APRS_iGate
> I (4475) cpu_start: App version:      v0.1-18-g9bb1908-dirty
> I (4479) cpu_start: Compile time:     Sep 25 2020 14:08:20
> I (4491) cpu_start: ELF file SHA256:  631741fb64855f9a...
> I (4495) cpu_start: ESP-IDF:          3.40100.200827
> I (4504) cpu_start: Starting app cpu, entry point is 0x40081030
> I (14686) cpu_start: App cpu up.
> I (28354) heap_init: Initializing. RAM available for dynamic allocation:
> I (28467) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (28520) heap_init: At 3FFB2980 len 0002D680 (181 KiB): DRAM
> I (28538) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (28595) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (28618) heap_init: At 4008A038 len 00015FC8 (87 KiB): IRAM
> I (28749) cpu_start: Pro cpu start user code
> I (7945) spi_flash: detected chip: gd
> I (8103) spi_flash: flash io: dio
> I (8174) cpu_start: Starting scheduler on PRO CPU.
> I (150) cpu_start: Starting scheduler on APP CPU.
> I (8831) example: Initializing SPIFFS
> W (9661) SPIFFS: mount failed, -10025. formatting...
> E (56221) SPIFFS: mount failed, -10025
> E (56241) example: Failed to mount or format filesystem

peterus, 2020-09-28:
after a some days of trying around and getting familiar to ESP-IDF I decided to switch to ESP-IDF.

closed 2020-09-28 by peterus