espressif/qemu #12: LoadStorePIFAddrError

caipiblack, 2020-11-20:
Hi

I am using qemu to emulate and debug my esp32 software but I have a problem: When I call the function "ledc_timer_config()", the execution run into an exception "LoadStorePIFAddrError"

**Example:**

```
        ledc_timer_config_t ledc_timer =
        {
            .duty_resolution = LEDC_TIMER_10_BIT,
            .freq_hz = 5000,
            .speed_mode = LEDC_LOW_SPEED_MODE,
            .timer_num = LEDC_TIMER_0
        };
        ledc_timer_config(&ledc_timer);
```

**Output:**

```
Guru Meditation Error: Core  0 panic'ed (LoadStorePIFAddrError). Exception was unhandled.
Core 0 register dump:
PC      : 0x401e4470  PS      : 0x00060833  A0      : 0x80150fb2  A1      : 0x3ffbb500
A2      : 0x3ffda2e8  A3      : 0xfffffffd  A4      : 0x00000fa0  A5      : 0x00000000
A6      : 0x00000002  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ff59000
A10     : 0xfffffffe  A11     : 0x00060820  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x0000cdcd  A15     : 0xfffffffe  SAR     : 0x0000000b  EXCCAUSE: 0x0000000f
EXCVADDR: 0x3ff59190  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

ELF file SHA256: cc9b36e22e3549cf

Backtrace: 0x401e446d:0x3ffbb500 0x40150faf:0x3ffbb520 0x40151173:0x3ffbb550 0x40103034:0x3ffbb590 0x400f7360:0x3ffbb5e0 0x400f7237:0x3ffbb650 0x400d7ab7:0x3ffbb670 0x400d406b:0x3ffbb6c0

CPU halted.
```

**Callstack:**

[image]

**Questions:**

Did you know what is the problem ? Something not implemented in qemu ?

Thanks for help

igrr, 2020-11-20:
Hi @caipiblack, indeed the LEDC peripheral is not emulated in QEMU. The memory region where LEDC peripheral is hasn't been defined in QEMU, so access to its address results in an exception.

You can silence this error by adding a dummy unimplemented device:
```c++
esp32_soc_add_unimp_device(sys_mem, "esp32.i2c1", DR_REG_LEDC_BASE, 0x1000);
```
next to the other lines here:
https://github.com/espressif/qemu/blob/0ff3da8d3c797dcf33a45c419204f39f684376cf/hw/xtensa/esp32.c#L452-L453

caipiblack, 2020-11-27:
Thanks,

Problem solved

closed 2020-11-27 by caipiblack