espressif/qemu #13: Control UART from QEMU

caipiblack, 2020-11-27:
Dear sir,

When I try to get the output of the UART's of the ESP32, i can only get the results from the first one.

Example using TCP Server:
```
qemu/xtensa-softmmu/qemu-system-xtensa -nographic -s -S -machine esp32 -drive file=final.bin,if=mtd,format=raw -nic user,model=open_eth -serial tcp::8880,server,nowait -serial tcp::8881,server,nowait
```

Example using files:
```
qemu/xtensa-softmmu/qemu-system-xtensa -nographic -s -S -machine esp32 -drive file=final.bin,if=mtd,format=raw -nic user,model=open_eth -serial file:/tmp/uart0.txt -serial file:/tmp/uart1.txt
```

Did we have to implement somthing in esp32.c ?

I try to add this line without success: 

```
object_property_add_alias(obj, "serial1", OBJECT(&s->uart[1]), "chardev", &error_abort);
```

caipiblack, 2020-11-30:
I found that i can add UART 1 & 2 by doing these modifications:

in the function “esp32_soc_init()”(esp32.c), add these lines:

```
object_property_add_alias(obj, "serial1", OBJECT(&s->uart[1]), "chardev", &error_abort);
object_property_add_alias(obj, "serial2", OBJECT(&s->uart[2]), "chardev", &error_abort);
```

in the function “esp32_machine_inst_init()”(esp32.c), add these lines:

```
qdev_prop_set_chr(DEVICE(s), "serial1", serial_hd(1));
qdev_prop_set_chr(DEVICE(s), "serial2", serial_hd(2));
```

Then to get the output of all serial lines, i use one of these lines:

`qemu/xtensa-softmmu/qemu-system-xtensa -nographic -s -S -machine esp32 -drive file=final.bin,if=mtd,format=raw -nic user,model=open_eth -serial tcp::8880,server,nowait -serial /dev/ttyS5 -serial tcp::8882,server,nowait`

`qemu/xtensa-softmmu/qemu-system-xtensa -nographic -s -S -machine esp32 -drive file=final.bin,if=mtd,format=raw -nic user,model=open_eth -serial file:/tmp/uart0.txt -serial file:/tmp/uart1.txt -serial file:/tmp/uart2.txt`

`qemu/xtensa-softmmu/qemu-system-xtensa -nographic -s -S -machine esp32 -drive file=final.bin,if=mtd,format=raw -nic user,model=open_eth -serial /dev/ttyS4 -serial /dev/ttyS5 -serial /dev/ttyS6`

I arrive to get the data transmit by the emulated ESP32, but I don't arrive to transmit data to the ESP32.

Did you know if we have something to change

caipiblack, 2020-12-03:
I got a response about it on the esp32 forum:

https://www.esp32.com/viewtopic.php?f=2&t=18358

But i don't know how to implement this on QEMU, did you have any ideas?

igrr, 2021-02-20:
@caipiblack RXFIFO_TOUT interrupt is implemented in the latest esp-develop branch. I have also included your suggestion for exposing UART1 and UART2, thanks for that!

closed 2021-02-20 by igrr

manimathma, 2022-07-26:
@caipiblack , couldn't get this to work. Do we need to call some init function from main?