espressif/qemu #17: Using flash image more that 4Mb works with errors

vvsosed, 2021-01-13:
In my work I use 16Mb spi flash and respectively I want to use the same image in qemu. I created flash image of 16M size and tried to use it in qemu. There are two problem was noticed. First of all I found that when I run such image it fails in bootloader code because of error:
E (10154) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.
At first look this difficulty can be overcome using CONFIG_SPI_FLASH_USE_LEGACY_IMPL=y parameter but I found that it isn't. I founded that if we writing in spi flash by adresses more than 0xFFFFFFFF (maximal uin32 value) it leads to damage of spi flash by the adresses lower than 0xFFFFFFFF. It seems to me that somewere in qemu source code 32bit values are using to specify the adresses in flash space and it leads to rounding error if we use flash images more than 4Mb. It is possible to use more than 4Mb flash in qemu and how?

fabian-schlegel, 2022-01-06:
Hey hey!

I found a handy solution for your problem.

Go to the file `hw/block/m25p80.c` and search for the line `/* GigaDevice */`. Change the block below this line, so it looks like:

```
/* GigaDevice */
{ INFO("gd25q32",     0xc84016,      0,  64 << 10,  64, ER_4K) },
{ INFO("gd25q64",     0xc84017,      0,  64 << 10, 128, ER_4K) },
{ INFO("gd25q128",    0xc84018,      0,  64 << 10, 256, ER_4K) },
```

After doing so, go to the file `hw/xtensa/esp32.c` and search for the line `DeviceState *flash_dev = qdev_new("gd25q32");`. Now you need to replace `gd25q32` (which sets the 4MB flash chip), to the desired flash size. For 8MB it is `gd25q64` and for 16MB it is `gd25q128`.

Now rebuild the project as described in the Wiki and enjoy your 4MB / 8MB / 16MB emulated flash...

I'll create a pull request to enable selecting the flash size as soon as I find some time...

max1220, 2022-10-02:
Any update on this? Could the flash chip be made selectable? I'd really like to test a larger application.

igrr, 2023-02-24:
Fix is being discussed in https://github.com/espressif/qemu/pull/68.