espressif/qemu #20: Support for esp8266?

raspiduino, 2021-07-19:
Could you add support for esp8266? Thanks!

[This](https://github.com/OSLL/qemu-xtensa/tree/xtensa-esp8266/hw/xtensa/) fork of QEMU 2.x provide (some kind of) esp8266 emulator, but I tried it and it didn't work with the serial Hello World test (missing peripherals or something?).

igrr, 2021-07-26:
This is technically possible. However ESP8266 emulation is unlikely to get a lot of use within Espressif (for automated testing or CI). So the emulation might get broken and might not be updated along with the changes to ESP8266_RTOS_SDK. In the end it might see the same fate as https://github.com/OSLL/qemu-xtensa/commits/xtensa-esp8266. Unless some actively maintained project (like https://github.com/esp8266/arduino) is interested in using it, I don't have confidence that this won't be a wasted effort.

svenstaro, 2022-04-25:
I suppose this should be closed as obviously there's little interest in getting this off the ground from either upstream or users.