espressif/qemu #21: The latest code compilation error, can not find the relevant header file.

ouyangkan, 2021-09-09:
I want to build qEMu with esp32 support and follow the instructions in readMe:  
mkdir build  
cd build  
.  /configure  
make  
Missing header file displayed during compilation,  

[image]

Excuse me how should I solve this problem, be to lack what to rely on.

ouyangkan, 2021-09-09:
I used ubuntu18.04。

ouyangkan, 2021-09-09:
First, I modified subprojects\libvhost-user\libvhost-user.h and subprojects\libvhost-user\include\atomic.h. 

Content as follows:
`#include "../../include/standard-headers/linux/virtio_ring.h"`
`#include "../../../include/qemu/atomic.h"`

igrr, 2021-09-09:
Please check https://github.com/espressif/qemu/wiki#configure, you need to pass `--enable-gcrypt` option to the configure script.

ouyangkan, 2021-09-09:
Thank you for your reply. Now my problem has been solved.

closed 2021-09-09 by ouyangkan