espressif/qemu #22: Open Ethernet emulation for esp-idf 3.2

createuniverses, 2021-09-13:
Many popular libraries are built on version 1.0.4 of the espressif/arduino-esp32 layer. This layer compiles against esp-idf 3.2, which does not include open ethernet. Unfortunately, this version of esp-idf does not include the open-eth component, so emulation of applications that are built on it can't do networking.

Can the open-eth component be included on the application side? This will allow applications built on esp-idf 3.2 to use the network features of the emulator.

In the newer versions of esp-idf that include open-eth, open-eth refers to some of the other components, so it is unclear if it can be cleanly separated from the esp-idf it is embedded in for use as a standalone component.

igrr, 2021-09-13:
This doesn't look easy to achieve as the Ethernet driver in IDF started supporting multiple MAC implementations only in v4.0. Until then there was only an implementation for the real ESP32 Ethernet controller. I guess you could try to override the entire component from your application. The implementation of openeth MAC driver would indeed need to be changed to match the Ethernet driver in v3.2.

svenstaro, 2022-04-25:
Isn't this now fixed with OpenEth in QEMU?

igrr, 2022-08-03:
I'll close this issue since IDF v3.2 is now EoL.

closed 2022-08-03 by igrr