espressif/qemu #24: Floating point exception (core dumped)

JiaDuo, 2021-12-09:
I test the "uart_echo" example. But failed. It show "Floating point exception (core dumped)"

CMD LOG:

$cd peripherals/uart/uart_echo
$idf.py build
$cd build
$esptool.py --chip esp32 merge_bin --fill-flash-size 4MB -o flash_image.bin 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 uart_echo.bin
$~/other/espressif_qemu/qemu/bin/qemu-system-xtensa -nographic -M esp32 -drive file=flash_image.bin,if=mtd,format=raw

QEMU OUTPUT:

Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7064
load:0x40078000,len:14308
load:0x40080400,len:3716
entry 0x40080680
I (888) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (894) boot: compile time 15:19:40
I (904) boot: chip revision: 0
I (912) boot.esp32: SPI Speed      : 40MHz
I (912) boot.esp32: SPI Mode       : DIO
I (913) boot.esp32: SPI Flash Size : 2MB
I (924) boot: Enabling RNG early entropy source...
I (941) boot: Partition Table:
I (941) boot: ## Label            Usage          Type ST Offset   Length
I (942) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (945) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (946) boot:  2 factory          factory app      00 00 00010000 00100000
I (950) boot: End of partition table
I (960) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08764h ( 34660) map
I (1016) esp_image: segment 1: paddr=0001878c vaddr=3ffb0000 size=0290ch ( 10508) load
I (1046) esp_image: segment 2: paddr=0001b0a0 vaddr=40080000 size=04f78h ( 20344) load
I (1065) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=165c4h ( 91588) map
I (1226) esp_image: segment 4: paddr=000365ec vaddr=40084f78 size=06a88h ( 27272) load
I (1271) esp_image: segment 5: paddr=0003d07c vaddr=50000000 size=00010h (    16) load
I (1297) boot: Loaded app from partition at offset 0x10000
I (1299) boot: Disabling RNG early entropy source...
I (1313) cpu_start: Pro cpu up.
I (1314) cpu_start: Starting app cpu, entry point is 0x400810e0
I (5429) cpu_start: App cpu up.
I (2247) cpu_start: Pro cpu start user code
I (2248) cpu_start: cpu freq: 160000000
I (2248) cpu_start: Application information:
I (2249) cpu_start: Project name:     uart_echo
I (2249) cpu_start: App version:      v4.3.1
I (2250) cpu_start: Compile time:     Dec  9 2021 15:19:35
I (2251) cpu_start: ELF file SHA256:  29985d43c52312e9...
I (2252) cpu_start: ESP-IDF:          v4.3.1
I (2259) heap_init: Initializing. RAM available for dynamic allocation:
I (2263) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (2265) heap_init: At 3FFB31D0 len 0002CE30 (179 KiB): DRAM
I (2266) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (2267) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2269) heap_init: At 4008BA00 len 00014600 (81 KiB): IRAM
I (2350) spi_flash: detected chip: gd
I (2375) spi_flash: flash io: dio
W (2395) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (2427) cpu_start: Starting scheduler on PRO CPU.
I (10) cpu_start: Starting scheduler on APP CPU.
Floating point exception (core dumped)

JiaDuo, 2021-12-09:
I found this is cause by '/hw/char/esp32_uart.c' . in fuction 'uart_set_rx_timeout'

`int64_t rx_timeout_ns = now + threshold_reg * 8 * NANOSECONDS_PER_SECOND / s->baud_rate;`

s->baud_rate is 'zer0' when run .

I do the fix 
```
        s->baud_rate = s->baud_rate ? s->baud_rate : 115200;

        int64_t rx_timeout_ns = now + threshold_reg * 8 * NANOSECONDS_PER_SECOND / s->baud_rate;
```

igrr, 2021-12-09:
Thanks for finding the issue and its cause @JiaDuo. I'll add this fix to the next release.

igrr, 2021-12-16:
The root cause was that only UART0 was properly reset and initialized here:
https://github.com/espressif/qemu/blob/fd85235d17cd8813d6a31f5ced3c5acbf1933718/hw/xtensa/esp32.c#L169-L171
this should have been `&s->uart[i]` not `&s->uart`.

igrr, 2022-02-03:
This is now fixed in https://github.com/espressif/qemu/releases/tag/esp-develop-20220203.

closed 2022-02-03 by igrr