espressif/qemu #26: Can't run code when using IDF v4.4

thinksilicon, 2021-12-20:
I've successfully used the emulator for running some PlatformIO/Arduino sketches. PlatformIO used IDF v3.3.5.

Recently I switched to IDF v4.4 and ever since the application doesn't run in the emulator anymore. It does run fine when I flash it to the actual hardware, so it looks like the code that's generated runs fine, generally speaking.

I'm running qemu with the following options:
```bin/qemu-system-xtensa -nographic -machine esp32 -drive file=../flash_image.bin,if=mtd,format=raw -global driver=esp32.gpio,property=strap_mode,value=0x12 -drive file=../efuse/qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32.efuse,property=drive,value=efuse```

The output I'm getting is
```
==22289==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
ets ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
```

It basically runs into a watchdog timeout. If I disable the watchdog, nothing happens at all.

Please let me know what other useful information I can provide to debug the problem. I upload the firmware through esptool.py and setting the esp32.gpio.property value to 0x01 and writing to the socket:

```
esptool.py v3.1
Serial port socket://localhost:5555
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000bafff...
Compressed 17216 bytes to 11867...
Writing at 0x00001000... (100 %)
Wrote 17216 bytes (11867 compressed) at 0x00001000 in 0.6 seconds (effective 229.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 611.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 627.1 kbit/s)...
Hash of data verified.
Compressed 698432 bytes to 445097...
Writing at 0x00010000... (3 %)
Writing at 0x0001d172... (7 %)
Writing at 0x0002a21e... (10 %)
[...]
Writing at 0x000b3f48... (96 %)
Writing at 0x000b9890... (100 %)
Wrote 698432 bytes (445097 compressed) at 0x00010000 in 21.0 seconds (effective 266.5 kbit/s)...
Hash of data verified.

Leaving...
```

igrr, 2021-12-20:
Thanks for reporting this @thinksilicon. I was able to reproduce the issue, but the cause of this issue is not clear yet. It seems that the startup code on CPU1 doesn't run properly in emulation. This might be connected to the fact that we are using CPU0 ROM binary when emulating CPU1. Unfortunately I can't give you any timeline for resolving this issue, but i'll update it if there is any progress.

thinksilicon, 2021-12-21:
Thanks for verifying! At least I know now it is an issue with the emulator and not my code.

igrr, 2022-01-02:
The issue was fixed in the `master` branch of esp-idf in https://github.com/espressif/esp-idf/commit/5d32e80ea7533a68447ed75d75277a8a80c7aa0d, we are also backporting it into release/v4.4 branch. Once that is done, arduino-esp32 will pick up this fix.

thinksilicon, 2022-01-06:
Cool, thank you! I'll try once the changes have made it into arduino-esp32.

me-no-dev, 2022-01-18:
@thinksilicon changes are in :) please test!

thinksilicon, 2022-01-19:
Are you planning on a new release of the arduino-esp32 framework with the updated IDF as well?
https://github.com/espressif/arduino-esp32/releases

Looks like the changes are not integrated into there just yet? (I've been using the Arduino framework when I discovered the behavior of this bug regport).

me-no-dev, 2022-01-20:
Yes we will. Can't quote a day yet, but soon :)

igrr, 2022-04-01:
Arduino 2.0.3-rc1 is now available and should contain the fix.

closed 2022-04-01 by igrr