espressif/qemu #30: Can I use the binary compiled by Arduino IDE instead of ESP-IDF?

raspiduino, 2022-01-25:
Hello,
I know it's possible to run binary compiled with ESP-IDF using this fork of QEMU. And the [Arduino IDE's support for ESP32](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk) is based on the ESP-IDF framework.

So is it possible to run Arduino IDE's compiled binary on this QEMU? I tried the option `Export compiled binary` on Arduino IDE and then load the binary to QEMU [as described](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk), but it just output

```text
==20167==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
qemu-system-xtensa: failed to read the initial flash content
```

Thanks in advance!

igrr, 2022-01-25:
I think the binary exported by Arduino IDE only includes the application. You probably need to combine it with the bootloader and partition table binaries at correct offsets. The resulting file should be exactly 4 megabytes. Please check the wiki page for an example script which does this using dd commands.

raspiduino, 2022-01-25:
Can you provide me the link?

igrr, 2022-01-25:
https://github.com/espressif/qemu/wiki/Home/6d464e83db0285059f53a078d9a1d3ad5aed5672#build-test-app

raspiduino, 2022-01-25:
Hello,

Why the wiki I saw (by clicking to `Wiki` tab) and the Wiki link you sent me are so much different?
<br> The wiki I saw by clicking to `Wiki` tab:
[image]
<br> The wiki you sent me:
[image]
<br> (I wonder if they are two version (old and new) of the wiki?)
<hr>
OK let's come to the main problem:

After follow the instructions in the wiki, QEMU does not print out `Hello World` but instead stuck at
```text
user@machine:~$ qemu-system-xtensa -nographic -machine esp32 -drive file=esp32.qemu.bin,if=mtd,format=raw
==20376==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
```
and then nothing else print out!

Here are my steps:
1) I use a simple piece of code that just print out `Hello World` to UART:
```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello World");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Board configure in Arduino IDE:
<br>[image]

Then I built it by using `Export compiled binary`:
<br>
[image]

Build log:
```text
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\USER\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\USER\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\USER\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=dout,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path C:\Users\USER\AppData\Local\Temp\arduino_build_60522 -warnings=none -build-cache C:\Users\USER\AppData\Local\Temp\arduino_cache_876638 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\USER\Documents\Arduino\esp32test\esp32test.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\USER\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\USER\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\USER\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=dout,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path C:\Users\USER\AppData\Local\Temp\arduino_build_60522 -warnings=none -build-cache C:\Users\USER\AppData\Local\Temp\arduino_cache_876638 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\USER\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\USER\Documents\Arduino\esp32test\esp32test.ino
Using board 'esp32' from platform in folder: C:\Users\USER\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\USER\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6
[... cut ...]
user@machine:~$ qemu-system-xtensa -nographic -machine esp32 -drive file=esp32.qemu.bin,if=mtd,format=raw
==20376==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
```

What am I doing wrong here? Thanks in advance!

igrr, 2022-01-25:
> Why the wiki I saw (by clicking to Wiki tab) and the Wiki link you sent me are so much different?

The latest version of the wiki was updated to use esptool.py for generation of flash binary image. This is easier than using `dd`,  in most cases. However since you are using Arduino IDE, you don't have the flash_args file (it's generated by IDF build system). So I gave you a link to an earlier version of the Wiki which had that `dd` based script.

It looks like you are using 1.0.6 version of arduino-esp32 core to test this. I suspect that something in the way that version of the core works with UART interface is not compatible with QEMU. It seems that arduino-esp32 version 2.0.x fixes this issue.
However there's currently another issue with arduino-esp32 reported in https://github.com/espressif/qemu/issues/26. It's already fixed in the master branch of arduino-esp32, but not part of any release yet. If you try building the sketch with the master branch of arduino-esp32, i think it should work. Alternatively, wait for the next arduino-esp32 release which will include the above mentioned fix.

raspiduino, 2022-01-25:
Ok I will try to clone the latest source in arduino-esp32 repo and try again.

Thanks for the information and I will post a comment here when I have something new.

raspiduino, 2022-01-26:
Okay I switched to Ubuntu (not WSL anymore) and install latest `arduino-esp32` board from [package_esp32_dev_index.json](https://github.com/espressif/arduino-esp32/releases/tag/2.0.2).

Board setting:
[image]

After using `Export compiled binary`, it showed some log. But it's too long for pasting here so I attached as [arduinoide.txt](https://github.com/espressif/qemu/files/7938781/arduinoide.txt)

In the directory `/tmp/arduino_build_171810/`:
```text
user@ubuntu:/tmp/arduino_build_171810$ ls
build_opt.h          hello_world.ino.bootloader.bin  includes.cache  sketch
build.options.json   hello_world.ino.elf             libraries
core                 hello_world.ino.map             partitions.csv
hello_world.ino.bin  hello_world.ino.partitions.bin  preproc
user@ubuntu:/tmp/arduino_build_171810$
```
I modified the script `make-flash-img.sh` to use for this purpose:
```text
user@ubuntu:/tmp/arduino_build_171810$ cat make-flash-img.sh 
#!/usr/bin/env bash
set -e
arg_projname=$1
arg_flashimg=$2

if [ -z "$1" -o -z "$2" ]; then
    echo "Usage: make-flash-img.sh app_name flash_img_file"
    exit 1
fi

dd if=/dev/zero bs=1024 count=4096 of=${arg_flashimg}
dd if=${arg_projname}.bootloader.bin bs=1 seek=$((0x1000)) of=${arg_flashimg} conv=notrunc
dd if=${arg_projname}.partitions.bin bs=1 seek=$((0x8000)) of=${arg_flashimg} conv=notrunc
dd if=${arg_projname}.bin bs=1 seek=$((0x10000)) of=${arg_flashimg} conv=notrunc
user@ubuntu:/tmp/arduino_build_171810$ ./make-flash-img.sh hello_world.ino hello_world.bin
4096+0 records in
4096+0 records out
4194304 bytes (4,2 MB, 4,0 MiB) copied, 0,0680444 s, 61,6 MB/s
17248+0 records in
17248+0 records out
17248 bytes (17 kB, 17 KiB) copied, 0,236782 s, 72,8 kB/s
3072+0 records in
3072+0 records out
3072 bytes (3,1 kB, 3,0 KiB) copied, 0,0411267 s, 74,7 kB/s
228432+0 records in
228432+0 records out
228432 bytes (228 kB, 223 KiB) copied, 2,08783 s, 109 kB/s
user@ubuntu:/tmp/arduino_build_171810$ ls
build_opt.h         hello_world.bin                 hello_world.ino.elf             includes.cache     partitions.csv
build.options.json  hello_world.ino.bin             hello_world.ino.map             libraries          preproc
core                hello_world.ino.bootloader.bin  hello_world.ino.partitions.bin  make-flash-img.sh  sketch
user@ubuntu:/tmp/arduino_build_171810$ 
```
Run it with QEMU:
```text
user@ubuntu:/tmp/arduino_build_171810$ qemu-system-xtensa -nographic -machine esp32 -drive file=hello_world.bin,if=mtd,format=raw
==23286==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[... 161 lines cut ...]
```text
rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
E (99124) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (99124) task_wdt:  - IDLE (CPU 0)
E (99124) task_wdt: Tasks currently running:
E (99124) task_wdt: CPU 0: main
E (99124) task_wdt: CPU 1: ipc1
E (99124) task_wdt: Aborting.

abort() was called at PC 0x400d7091 on core 0

Backtrace:0x40082dfd:0x3ffbe7800x400879f1:0x3ffbe7a0 0x4008c2f9:0x3ffbe7c0 0x400d7091:0x3ffbe840 0x40084079:0x3ffbe860 0x400867f4:0x3ffbd210 0x4008682c:0x3ffbd230 0x4008694c:0x3ffbd250 0x400d1dfc:0x3ffbd270 0x400d16d2:0x3ffbd2b0 0x400d1644:0x3ffbd2d0 0x400ebf30:0x3ffbd300 

ELF file SHA256: 0000000000000000

Rebooting...
```
interesting for you?

raspiduino, 2022-01-26:
After setting the option `Core Debug Version` to `Verbose`, I got:
```text
user@ubuntu:/tmp/arduino_build_171810$ qemu-system-xtensa -nographic -machine esp32 -drive file=hello_world.bin,if=mtd,format=raw
==23740==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[   930][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[  1167][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 1ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[  1035][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[  1225][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (30581) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (30581) task_wdt:  - IDLE (CPU 0)
E (30581) task_wdt: Tasks currently running:
E (30581) task_wdt: CPU 0: main
E (30581) task_wdt: CPU 1: ipc1
E (30581) task_wdt: Aborting.

abort() was called at PC 0x400d7445 on core 0

Backtrace:0x40082dfd:0x3ffbe7800x400879f1:0x3ffbe7a0 0x4008c2f9:0x3ffbe7c0 0x400d7445:0x3ffbe840 0x40084079:0x3ffbe860 0x400d21b3:0x3ffbd250 0x400d178a:0x3ffbd2b0 0x400d16fc:0x3ffbd2d0 0x400ec31c:0x3ffbd300 

ELF file SHA256: 0000000000000000

Rebooting...
```

igrr, 2022-01-26:
@raspiduino thanks for trying this out, sorry it didn't work as expected! Seems like you are using version 2.0.2 which still has the issue reported in #26. It is fixed in arduino-esp32 master branch (installable via git) but is not in any release yet.
I'll try this with the `master` branch of arduino-esp32 and let you know whether it actually works in QEMU...

raspiduino, 2022-01-26:
I tried the latest source from git repo, but it still:
```
user@ubuntu:/tmp/arduino_build_130815$ qemu-system-xtensa -nographic -machine esp32 -drive file=hello_world.bin,if=mtd,format=raw
==54624==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[  3047][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[  3426][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[  2959][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[  3217][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[  3140][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (116719) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (116719) task_wdt:  - IDLE (CPU 0)
E (116719) task_wdt: Tasks currently running:
E (116719) task_wdt: CPU 0: main
E (116719) task_wdt: CPU 1: ipc1
E (116719) task_wdt: Aborting.

abort() was called at PC 0x400d778d on core 0

Backtrace:0x40082e21:0x3ffbe79c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

raspiduino, 2022-01-27:
Hello,
From #26 thinksilicon [said that](https://github.com/espressif/qemu/issues/26#issue-1084369691):
> I've successfully used the emulator for running some PlatformIO/Arduino sketches. PlatformIO used IDF v3.3.5.

So does that mean I can go back to IDF 3.3.5 as a temporary fix?

igrr, 2022-08-03:
I have verified that QEMU can be used to emulate binaries produced with Arduino IDE, using arduino-esp32 core v2.0.4.

1. Compiled the application for ESP32 Dev module, 4MB flash size, DIO flash mode.
2. "Export compiled binary" from the Sketch menu of the IDE
3. Go to the directory where the sketch was built (should be visible in the log output) and create a merged binary using esptool (modify the command to match your sketch name):
  ```
  esptool.py --chip esp32 \
    merge_bin --flash_mode dio --flash_size 4MB --fill-flash-size 4MB --output merged.bin \
    0x1000 test_qemu.ino.bootloader.bin \
    0x8000 test_qemu.ino.partitions.bin \
    0x10000 test_qemu.ino.bin
  ```
4. Run QEMU:
  ```
  qemu-system-xtensa -nographic -machine esp32 \
    -drive file=merged.bin,if=mtd,format=raw \
    -global driver=timer.esp32.timg,property=wdt_disable,value=true
  ```
  
The final line is important, we have to disable the Timer Group watchdog emulation, otherwise it will trigger during the boot process. This is because the boot process for an Arduino sketch is notably slower in QEMU than on a real chip. Note that there is also some delay between `setup` and `loop` execution, I haven't looked into this further. After some time both `setup` and `loop` output should be visible.

closed 2022-08-03 by igrr

raspiduino, 2022-08-03:
Thanks! I will try this as soon as I can.