espressif/qemu #32: Watchdog timer fires when trying to use SPIFFS

smuehlst, 2022-02-01:
I'm trying to use SPIFFS under QEMU in the same manner as I'm already doing it in a program that runs successfully on a Tiny S2 board. When I run the program in QEMU, the watchdog timer fires:

```
...
I (10453) example: Initializing SPIFFS
E (15303) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15303) task_wdt:  - IDLE (CPU 0)
E (15303) task_wdt: Tasks currently running:
E (15303) task_wdt: CPU 0: main
E (15303) task_wdt: CPU 1: IDLE
E (15303) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40184013:0x3FFB0840 0x40084D95:0x3FFB0860 0x40086A64:0x3FFCDC60 0x40085867:0x3FFCDC80 0x400867E5:0x3FFCDCA0 0x40086892:0x3FFCDCC0 0x400872B5:0x3FFCDCE0 0x4008664A:0x3FFCDD00 0x4017FB29:0x3FFCDD40 0x400E0A25:0x3FFCDD70 0x400E4965:0x3FFCDDA0 0x400E25CA:0x3FFCDDD0 0x400E1009:0x3FFCDE10 0x400E00D4:0x3FFCDE40 0x400E098F:0x3FFCDE80 0x400DB77A:0x3FFCDF50 0x400DBA0D:0x3FFCDFA0 0x401A30E7:0x3FFCE020 0x4008E591:0x3FFCE040

E (15303) task_wdt: Print CPU 1 backtrace

Backtrace:0x40087201:0x3FFB0E40 0x40084D95:0x3FFB0E60 0x4000BFED:0x3FFCF440 0x4008E849:0x3FFCF450 0x401842DF:0x3FFCF470 0x401842EB:0x3FFCF4A0 0x400DAF05:0x3FFCF4C0 0x4008CE6D:0x3FFCF4E0 0x4008E591:0x3FFCF500
```

By using the `xtensa-esp32s3-elf-addr2line` utility I was able to decode the addresses to the following stack trace:

```
0x40184013: task_wdt_isr at E:/Users/stm/Downloads/esp-idf/components/esp_common/src/task_wdt.c:189
0x40084d95: _xt_lowint1 at E:/Users/stm/Downloads/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x40086a64: xt_int_enable_mask at E:/Users/stm/Downloads/esp-idf/components/xtensa/include/xtensa/xtensa_api.h:170
 (inlined by) intr_cntrl_ll_enable_int_mask at E:/Users/stm/Downloads/esp-idf/components/hal/esp32/include/hal/interrupt_controller_ll.h:100
 (inlined by) interrupt_controller_hal_enable_int_mask at E:/Users/stm/Downloads/esp-idf/components/hal/include/hal/interrupt_controller_hal.h:192
 (inlined by) esp_intr_noniram_enable at E:/Users/stm/Downloads/esp-idf/components/esp_system/intr_alloc.c:815
0x40085867: spi_flash_enable_interrupts_caches_and_other_cpu at E:/Users/stm/Downloads/esp-idf/components/spi_flash/cache_utils.c:204
0x400867e5: cache_enable at E:/Users/stm/Downloads/esp-idf/components/spi_flash/spi_flash_os_func_app.c:63
0x40086892: spi1_end at E:/Users/stm/Downloads/esp-idf/components/spi_flash/spi_flash_os_func_app.c:111
0x400872b5: spiflash_end_default at E:/Users/stm/Downloads/esp-idf/components/spi_flash/esp_flash_api.c:125
0x4008664a: esp_flash_read at E:/Users/stm/Downloads/esp-idf/components/spi_flash/esp_flash_api.c:681
0x4017fb29: esp_partition_read at E:/Users/stm/Downloads/esp-idf/components/spi_flash/partition.c:419
0x400e0a25: spiffs_api_read at E:/Users/stm/Downloads/esp-idf/components/spiffs/spiffs_api.c:36
0x400e4965: spiffs_phys_rd at E:/Users/stm/Downloads/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:146
0x400e25ca: spiffs_obj_lu_scan at E:/Users/stm/Downloads/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:366 (discriminator 4)
0x400e1009: SPIFFS_mount at E:/Users/stm/Downloads/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:134
0x400e00d4: esp_spiffs_init at E:/Users/stm/Downloads/esp-idf/components/spiffs/esp_spiffs.c:246
0x400e098f: esp_vfs_spiffs_register at E:/Users/stm/Downloads/esp-idf/components/spiffs/esp_spiffs.c:379
0x400db77a: app_init_spiffs at E:\Users\stm\Documents\GIT\esp32-pdflib\esp-qemu-sdcard\build/../main/sd_card_example_main.c:133
0x400dba0d: app_main at E:\Users\stm\Documents\GIT\esp32-pdflib\esp-qemu-sdcard\build/../main/sd_card_example_main.c:242 (discriminator 13)
0x401a30e7: main_task at E:/Users/stm/Downloads/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)
0x4008e591: vPortTaskWrapper at E:/Users/stm/Downloads/esp-idf/components/freertos/port/xtensa/port.c:168
```
Should SPIFFS work under QEMU? Are additional command line options for QEMU necessary?

I'm using ESP-IDF v4.3.2.

igrr, 2022-02-01:
Hi @smuehlst, I think SPI flash emulation in QEMU works slower than it does on the real chip. For this reason the same operation which doesn't trigger the task watchdog on the real chip may trigger it in QEMU.

There are several things you could do: disable watchdogs in QEMU (see wiki), increase task watchdog timeout in sdkconfig when building for QEMU, or change some sdkconfig options for yielding periodically during flash operations.

Ideally, SPI flash emulation speed in QEMU could be improved,.but I haven't found any obvious ways to do this, so far.

smuehlst, 2022-02-02:
Hi @igrr, thanks for your suggestions.

I had already tried to disable the watchdogs according to the instructions in the Wiki by passing the option `-global driver=timer.esp32.timg,property=wdt_disable,value=true`. In this case the watchdog does not fire, but the program apparently hangs in the SPIFFS initialisation.

This is my SPIFFS initialisation function that I have more or less copied from one of the ESP-IDF examples, and that works on a real ESP32 board:

```
#define SPIFFS_MOUNT_POINT "/spiffs"

static void app_init_spiffs(void)
{
    ESP_LOGI(TAG, "Initializing SPIFFS");

    esp_vfs_spiffs_conf_t const conf = {
      .base_path = SPIFFS_MOUNT_POINT,
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = false
    };

    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return;
    }

    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
    }
}
```

In the QEMU output I see `Initializing SPIFFS`, and then nothing happens anymore, although I have enabled all the SPIFFS debug options in the `sdkconfig` file except "Enable SPIFFS Filesystem Visualization".

Are maybe additional settings in the `sdkconfig` file necessary to make SPIFFS work under QEMU?

smuehlst, 2022-02-02:
@igrr I have to correct myself: It just takes a really long time to initialise SPIFFS, on my machine approx. 60 seconds, and then everything proceeds as expected. I was just too impatient, I'm sorry for that.

closed 2022-02-02 by smuehlst