espressif/qemu #33: Missing libraries in released executable

tore-espressif, 2022-02-21:
When I run [latest release](https://github.com/espressif/qemu/releases/tag/esp-develop-20220203) on fresh installation of Ubuntu 20.04 it fails to start. I must install missing libraries:
``` sh
apt install libpixman-1-0 libgio-cil
```

Are these libraries intentionally omitted from the release?

igrr, 2022-08-03:
Here's the list of packages installed by Ubuntu 20.04 as dependencies for qemu(-system-arm, but should be the same for Xtensa): https://packages.ubuntu.com/focal/qemu-system-arm.

`libpixman-1-0` is in that list, so I can assume that it's just a runtime dependency which has to be installed. I wasn't able to find the reason why libgio-cil has to be installed.

In general, installing the packages listed here should be enough to run QEMU: https://wiki.qemu.org/Hosts/Linux#Required_additional_packages. Some are only build dependencies, so probably could be omitted.

I think we'll try to solve this problem when creating an official Docker image, e.g. `espressif/idf-qemu` or similar.

igrr, 2022-08-04:
Having checked the internal qemu-v5.0 docker image, I see that we are installing:

    libpixman-1-0
    libnuma1
    libglib2.0-0

igrr, 2023-12-07:
QEMU along with its dependencies is now available in espressif/idf:latest Docker image.