espressif/qemu #35: MAC OSX Monterey build warnings

pkarashchenko, 2022-03-06:
Run build with macOS Monterey. Observe next warnings:
```
../block/file-posix.c:3335:18: warning: 'IOMasterPort' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
    kernResult = IOMasterPort( MACH_PORT_NULL, &masterPort );
                 ^~~~~~~~~~~~
                 IOMainPort
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:132:1: note: 'IOMasterPort' has been explicitly marked deprecated here
IOMasterPort( mach_port_t       bootstrapPort,
^
1 warning generated.
...
../audio/coreaudio.c:50:5: warning: 'kAudioObjectPropertyElementMaster' is deprecated: first deprecated in macOS 12.0 [-Wdep
recated-declarations]
    kAudioObjectPropertyElementMaster
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    kAudioObjectPropertyElementMain
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreAudio.framework/Headers/AudioHardwareBase.h:208:5: note: 'kAudioObjectPropertyElementMaster' has been explicitly marked deprecated here
    kAudioObjectPropertyElementMaster API_DEPRECATED_WITH_REPLACEMENT("kAudioObjectPropertyElementMain", macos(10.0, 12.0), ios(2.0, 15.0), watchos(1.0, 8.0), tvos(9.0, 15.0)) = kAudioObjectPropertyElementMain
    ^
../audio/coreaudio.c:72:9: warning: 'kAudioObjectPropertyElementMaster' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        kAudioObjectPropertyElementMaster
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        kAudioObjectPropertyElementMain
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreAudio.framework/Headers/AudioHardwareBase.h:208:5: note: 'kAudioObjectPropertyElementMaster' has been explicitly marked deprecated here
    kAudioObjectPropertyElementMaster API_DEPRECATED_WITH_REPLACEMENT("kAudioObjectPropertyElementMain", macos(10.0, 12.0), ios(2.0, 15.0), watchos(1.0, 8.0), tvos(9.0, 15.0)) = kAudioObjectPropertyElementMain
    ^
../audio/coreaudio.c:89:9: warning: 'kAudioObjectPropertyElementMaster' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        kAudioObjectPropertyElementMaster
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        kAudioObjectPropertyElementMain
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreAudio.framework/Headers/AudioHardwareBase.
h:208:5: note: 'kAudioObjectPropertyElementMaster' has been explicitly marked deprecated here
    kAudioObjectPropertyElementMaster API_DEPRECATED_WITH_REPLACEMENT("kAudioObjectPropertyElementMain", macos(10.0, 12.0),
ios(2.0, 15.0), watchos(1.0, 8.0), tvos(9.0, 15.0)) = kAudioObjectPropertyElementMain
    ^
6 warnings generated.
...
../ui/cocoa.m:1411:16: warning: 'setAllowedFileTypes:' is deprecated: first deprecated in macOS 12.0 - Use -allowedContentTy
pes instead [-Wdeprecated-declarations]
    [openPanel setAllowedFileTypes: supportedImageFileTypes];
               ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/AppKit.framework/Headers/NSSavePanel.h:215:49:
 note: property 'allowedFileTypes' is declared deprecated here
@property (nullable, copy) NSArray<NSString *> *allowedFileTypes API_DEPRECATED("Use -allowedContentTypes instead", macos(10
.3,12.0));
                                                ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/AppKit.framework/Headers/NSSavePanel.h:215:49:
 note: 'setAllowedFileTypes:' has been explicitly marked deprecated here
1 warning generated.
```

igrr, 2022-03-17:
I'm seeing these warnings in upstream release 6.2 as well. I recall seeing a patch set for macOS 12.0 support posted in January, so I suppose this will be fixed upstream in the next 7.0 release.

pkarashchenko, 2022-03-17:
Ok. We can either close this issue or keep it open until the esp32 version is rebased on top of upstream version with fix.

igrr, 2022-03-17:
Sure, I'll close this once 7.0 is released and our patches are rebased on top of it.

pkarashchenko, 2022-03-18:
The issue does not happen with upstream qemu

igrr, 2022-03-18:
Could you say, which upstream commit or tag have you used?

pkarashchenko, 2022-03-18:
`master` branch `commit 1d60bb4b14601e38ed17384277aa4c30c57925d3`

igrr, 2022-03-18:
Right, makes sense, earlier I was testing with the [v6.2.0 tag](https://github.com/qemu/qemu/tree/v6.2.0) which our branch is currently based on. Thanks for confirming that this issue is fixed in master. Once v7.0.0 is released (should be in the 2nd half of April), I'll rebase our branch and publish the new release.

igrr, 2022-08-02:
Sorry this took a lot longer than expected! https://github.com/espressif/qemu/releases/tag/esp-develop-20220802 release is based on the upstream v7.0.0 release and should fix the issue.

Note that I now get a different warning when building the upstream v7.0.0 release on macOS:
```
[963/2151] Compiling Objective-C object libcommon.fa.p/ui_cocoa.m.o
../ui/cocoa.m:801:17: warning: variable 'switched_to_fullscreen' set but not used [-Wunused-but-set-variable]
    static bool switched_to_fullscreen = false;
                ^
1 warning generated.
```

v7.1.0 is coming soon, hopefully it will fix that new warning.

closed 2022-08-02 by igrr