espressif/qemu #36: MAC OSX Monterey qemu-system-xtensa Library not loaded: @rpath/libclang_rt.asan_osx_dynamic.dylib

pkarashchenko, 2022-03-06:
maxOS Monterey (x86)
1. Configure build system with `../configure --target-list=xtensa-softmmu --enable-debug --enable-sanitizers --disable-strip --disable-capstone --disable-vnc --enable-gcrypt`
2. Run `make vga=no -j$(nproc)`
3. Run `make install`
4. Change directory and compile test project
5. Run test project with `qemu-system-xtensa -nographic -M esp32 -m 4 -drive file=flash.bin,if=mtd,format=raw -nic user,model=open_eth,hostfwd=tcp::80-:80 -s`

The next output is observed:
```
dyld[9082]: Library not loaded: @rpath/libclang_rt.asan_osx_dynamic.dylib
  Referenced from: /usr/local/bin/qemu-system-xtensa
  Reason: tried: '/usr/local/lib/libclang_rt.asan_osx_dynamic.dylib' (no such file), '/usr/lib/libclang_rt.asan_osx_dynamic.dylib' (no such file)
Abort trap: 6
```

Apply workaround: `../configure --target-list=xtensa-softmmu --enable-debug --enable-sanitizers --disable-strip --disable-capstone --disable-vnc --enable-gcrypt --extra-ldflags="-Wl,-rpath,/Library/Developer/CommandLineTools/usr/lib/clang/13.0.0/lib/darwin"`

igrr, 2022-03-17:
I haven't tried this, but based on the description it seems like this isn't an ESP-specific issue? I suppose it's better to report it upstream.

pkarashchenko, 2022-03-17:
Ok. I will try with the upstream and if it is reproduced, then I will close issue here.

pkarashchenko, 2022-03-18:
I can confirm that the issue is reproduced with the upstream

pkarashchenko, 2022-03-18:
I think that alternative method is described in https://wiki.qemu.org/Hosts/Mac "Build with LLVM/Clang 7" section. I will try this and close this issue if it works. Otherwise will submit issue to upstream

pkarashchenko, 2022-03-18:
It worked with
```
../configure --target-list=xtensa-softmmu --enable-debug --enable-sanitizers --disable-strip --disable-capstone --disable-vnc --enable-gcrypt --cc=clang-13 --cxx=clang++-13 --host-cc=clang-13 --extra-cflags="-I/usr/local/include" --extra-cxxflags="-I/usr/local/opt/llvm/include" --extra-ldflags="-L/usr/local/opt/llvm/lib -L/usr/local/opt/libffi/lib -L/usr/local/opt/llvm/lib -Wl,-rpath,/usr/local/opt/llvm/lib"
```
I had to specify `--extra-cflags="-I/usr/local/include"` additionally otherwise build fails unable to find `gcrypt.h` in `#include <gcrypt.h>`.
Anyway seems to be a usable workaround, so closing this issue.

closed 2022-03-18 by pkarashchenko