espressif/qemu #37: SPI/I2C exposed

HikingDev, 2022-03-20:
Hi,

i am trying to understand how to stimulate the Simulated Peripherals.
I do understand how to communicate via Sockets to the UART Peripheral.
Is there any way to use Sockets to communicate to SPI and/or I2C?

igrr, 2022-03-20:
Hi @HikingDev,
As far as I know QEMU doesn't support SPI/I2C forwarding via sockets. There is a number of SPI and I2C devices implemented in QEMU itself, such as certain temperature sensors and ADCs. In theory they can be used as described here: https://gist.github.com/jonte/b4bd83a5f2e8330418b1f3322bff74f2.

However this approach currently doesn't work with this ESP32 fork. It is because the SPI/I2C "buses" (QEMU term, no relation to physical buses) are not attached to the "default system bus". If a peripheral is attached to the default system bus in QEMU, it gets reset on system reset. On an ESP32 however there are multiple types of system reset, some of which do not reset all the peripherals. To emulate those reset types all peripherals got attached to a separate bus, unreachable from sysbus_default.

For SPI and I2C I think we can live with slightly imprecise emulation and have them reset on every type of system reset. That should make it possible to attach them to sysbus_default, and thus to attach emulated peripherals to SPI and I2C.
This still doesn't enable SPI or I2C communication over sockets, but it's a start.

HikingDev, 2022-03-21:
Hi @igrr ,
thank you for the quick response.
I am trying to read more about the "multiple types of system reset" as well as understanding the QEMU bus system.

Unfortunately the emulated SPI and I2C devices are not useful for our use case.

We use the esp32 as wifi module slave in Station Mode.
The net communicaiton happens via SPI whereas the commands (Scan, Connect etc) is done via UART.
So far the simulated host is able to update the firmware via UART on the emulated ESP32.
Of course it would be awesome if at some point the SPI protocol parallel to to already possible UART protocol could be developed and tested.

I I think that the exposed Devices are not a rare use case. Would you say that it is a rather complex task, exposing the aforementioned devices?

I guess i need to understand a little more of the internals of QEMU in general.

igrr, 2022-03-21:
I think there are three things which would have to be implemented:

* Support for SPI slave mode emulation. Currently only master mode is supported. It seems to be a bit tricky to support both, since in QEMU the device can't be both master and slave on the same bus, as far as I can tell. So we'll have to somehow split the SPI peripheral into two devices.
* Implement some kind of SPI master which would implement a character device type. This part wouldn't be ESP32 specific.
* Figure out how to attach ESP32 SPI slave peripheral to to this custom SPI master, if the user requests to create it.

If you are interested in implementing this, I can give you a few pointers around the codebase.

Unfortunately we can't commit to implementing this functionality ourselves. Medium term, our goals for QEMU will be related to improving emulation of security-related peripherals and adding emulation of new chips (ESP32-S2, ESP32-C3).

HikingDev, 2022-03-21:
This sounds rather time intensive. 
We are currently discussing if such a task could be formulated and amplified  that it is sufficient for bachelor thesis.
Unfortunately I doubt that I have and/or get the time capacity implement it (whereas I am intrigued to learn more about QEMU).

I will let you know if we somehow get the resources to implement a "QEMU SPI Slave"

TobiasNils, 2023-07-14:
> For SPI and I2C I think we can live with slightly imprecise emulation and have them reset on every type of system reset. That should make it possible to attach them to sysbus_default, and thus to attach emulated peripherals to SPI and I2C.

Hi @igrr , 
we are currently trying to figure out how to attach emulated peripherals to SPI. Would you be so kind and expand a little on how this "reset on every type of system reset" could be accomplished?

igrr, 2023-07-14:
Could you try replacing s->&periph_bus with NULL in https://github.com/espressif/qemu/blob/f0d55a072246e76d3b944db6374035b271d3da71/hw/xtensa/esp32.c#L454 and see if that helps?

lcgamboa, 2023-07-14:
Hi @TobiasNils,

in my fork I use [ssi_create_peripheral(s->spi[2].spi, "picsimlab_spi");](https://github.com/lcgamboa/qemu/blob/a959083356350fc5f05fd8752b8c74310adda892/hw/xtensa/esp32_picsimlab.c#L699C57-L699C57) to attach the [picsimlab_spi ](https://github.com/lcgamboa/qemu/blob/picsimlab-esp32/hw/ssi/picsimlab_spi.c) slave peripheral.  You can use this code as a base, just add the functionality you want to implement in the xxx_SPI_transfer function that is called every time ESP32 sends a byte and blocks qemu waiting for a return byte.

TobiasNils, 2023-07-17:
Thanks a lot @igrr and @lcgamboa. We will do some tests and report back.