espressif/qemu #41: Mounting memory onto host

nated0g, 2022-04-16:
First, this project is great and I've been having a lot of fun with it. 🙏

I'm trying to access the memory of the emulated ESP32 from my host system, and I was wondering if you'd had any experience with that.  Specifically, I'm trying to access the GPIO registers: 
```
memory-region: esp32.gpio
  000000003ff44000-000000003ff44fff (prio 0, i/o): esp32.gpio
```

I'm starting qemu with the following options in an attempt to mount those 4KB to a file

```
qemu-system-xtensa -nographic \
    -machine esp32 \
    -drive file=${PROJECT_BIN},if=mtd,format=raw \
-nic user,model=open_eth,id=lo0,hostfwd=tcp:0.0.0.0:8012-:80 \
-serial tcp::5555,server,nowait \
-monitor stdio \
-object memory-backend-file,id=esp32.gpio,size=4k,mem-path=./qemu-gpio,share=on \
-machine memory-backend=esp32.gpio
```

A file of the expected size is created, but it is blank, and does not update at all when running a blink sketch, for example.

nated0g, 2022-04-16:
Ah, it seems GPIO may not be implemented currently?

https://github.com/espressif/qemu/blob/440ff71648f0a6682ad59cf34929c0172c7f27af/hw/gpio/esp32_gpio.c#L39

igrr, 2022-04-19:
Hi @nated0g, I'm sorry, I didn't encounter `memory-backend` and `memory-backend-file` options until reading your question. I read the documentation and I think I understand how these options work for "regular" memory.

I still don't understand how the file memory backend interacts with MMIO devices in QEMU. Normally QEMU flattens the address space tree. If there are any overlapping memory regions, only one will handle read/write operation for a given address, and this will depend on the priority of each memory region. I suppose that the outcome you expect is that the write operations will be visible both in esp32.gpio device (which provides an MMIO region) and in the file-based memory backend. I don't know if QEMU actually supports this. If you already figured this part out, please let me know!

You can experiment with some other ESP32 peripheral which is implemented (like UART). Indeed, ESP32 GPIO emulation in QEMU isn't implemented, mostly since there aren't a lot of useful ways to let "gpio" devices in QEMU interact with the outside world. We'll consider implementing it, though.