espressif/qemu #42: esp32-s3 support

svenstaro, 2022-05-04:
I'd like to request support for esp32-s3 emulation. I think it's a killer chip and support for it in QEMU would be amazing.

cmorganBE, 2022-07-25:
Ping on this ticket. We'd like to be able to emulate s3 as that's our particular target.

DatanoiseTV, 2023-02-24:
I +1 this. I am rolling out hardware using the S3 and would love to see support for the S3, so I can perform some CI/CD tests.

igrr, 2023-02-24:
In the meantime, you can try https://wokwi.com/.

https://github.com/wokwi/esp32-hello-wokwi-ci should also be possible to adjust for the esp32-s3.

paul356, 2023-03-09:
+1 vote for this issue. Please support S3.

greenaddress, 2023-10-03:
+1. Please support S3

herculanodavi, 2024-01-28:
Hi, I'm also very interested in this. Is there any rough timeline?

drewbharris, 2024-03-11:
Adding my +1 please!

igrr, 2024-03-22:
Status update: we are aiming to add the initial ESP32-S3 support with a few basic peripherals in the upcoming release.

igrr, 2024-06-06:
Thanks everyone for the patience! Initial ESP32-S3 support has been released in https://github.com/espressif/qemu/releases/tag/esp-develop-9.0.0-20240606. We will update ESP-IDF integration (`idf.py qemu`) to be compatible with it soon.

closed 2024-06-06 by igrr

PatrickV82, 2024-06-14:
is there a updated documentation for the bootstrapping pins values?

Zitrax, 2024-08-09:
> We will update ESP-IDF integration (idf.py qemu) to be compatible with it soon.

Any estimate on this? Or info on how to run it without?

igrr, 2024-08-09:
@Zitrax This has been done in https://github.com/espressif/esp-idf/commit/6ddf31bbdcb7caf3817285cd732c4ce68097c034 and https://github.com/espressif/esp-idf/commit/aad9ab02e8200e079fb8b6e8e76268e04531d7b4

igrr, 2024-08-09:
@PatrickV82 Sorry that I missed your comment!

> is there a updated documentation for the bootstrapping pins values?

Not yet, but you can set `strap_mode` to 0x07 to enter download mode. Note that for efuses, you have to use `nvram.esp32c3.efuse` — this module was copy-pasted from ESP32-C3, we will fix this in the next QEMU release.