espressif/qemu #44: Missing RMT Peripheral

justdaniel-gh, 2022-06-02:
RMT's memory is not allocated in the machine so it panics when trying to initialize.

Just adding it as an unimplemented device (`esp32_soc_add_unimp_device(sys_mem, "esp32.rmt", DR_REG_RMT_BASE, 0x1000);`) seems to satisfy it.  I can submit a PR for it (just a one line change), but I'm curious what it would take to actually implement it. In my test case I don't really care about that feature (Just want it to not crash). I'm just using qemu to quickly test other functionality before flashing it to a board.

igrr, 2022-06-02:
Hi @justdaniel-gh,
Thanks for pointing this out! We'll be making a new release based on QEMU 7.0 soon, I will include this change. (You can of course submit a PR, but I will squash the commit when rebasing anyway, as we'd like to maintain ~one commit per feature in our fork.)

As for emulating the RMT, i guess the question is how to expose RMT inputs outside of QEMU, and how to RMT "inputs" into QEMU. QEMU doesn't have a framework for bitbang or similar interfaces, so we'd have to come up with something custom.

justdaniel-gh, 2022-06-03:
Unfortunately I don't have any direct experience working with virtualizing gpio, but it appears that some effort has gone in to it over the last couple of years.

The work described here looks promising: https://cs.unibo.it/~davide.berardi6/post/20201204-1.html

I'll try playing with something like this and see what I come up with.

igrr, 2022-09-19:
Unimplemented RMT device added in https://github.com/espressif/qemu/releases/tag/esp-develop-20220919.

closed 2022-09-19 by igrr