espressif/qemu #45: esp_flash support

timkoers, 2022-06-28:
I'd like to request support for the spi flash emulation, so that I can use esp_vfs_fat_spiflash_mount and all the other related functions from within my unit test

igrr, 2022-06-28:
@timkoers this should be already supported. Just keep in mind that the emulation performance is pretty low, since every operation is first represented as SPI commands (serialized to bytes) which are then handled by SPI Flash chip emulation layer. If you are running one of the flash related examples, you need to wait a fair bit, especially if the code is formatting a large partition.

timkoers, 2022-06-30:
The examples that I could find where of the SD MMC and the internal flash. I'd like to emulate an external SPI flash. Whilst your at it, would you mind providing an example on how to create such a image to be used during qemu emulation?

igrr, 2022-06-30:
Okay, I understand your use case now. Indeed there is no support for emulating external flash in QEMU. In principle this should be doable, need to create another instance of the SPI controller (SPI2) and attach a flash chip to it in the emulator.

I would suggest to try the emulation of the internal flash, first. It's pretty slow, so please make sure it is okay for your purposes before we try to add support for external flash.

Regarding creating the flash image of an app, you can follow the instructions in this wiki section: https://github.com/espressif/qemu/wiki#build-test-app

igrr, 2022-06-30:
You should be able to use the filesystem in the internal flash, please refer to storage/wear_levelling example in ESP-IDF.

igrr, 2022-08-03:
Closing this issue since support for esp_vfs_fat_spiflash_mount for the internal flash is currently available, and works in QEMU.

closed 2022-08-03 by igrr

timkoers, 2022-10-05:
I'd still like to see this getting implemented for an external flash since I don't want to mess around with the internal flash as that contains the actual program.