espressif/qemu #46: spi_flash_mmap not working

manimathma, 2022-07-06:
Dear Team and @igrr,

Thanks for the great platform you are providing us an ability to simulate and test our application. 

It seems like spi_flash_mmap doesn't seems to be working. spi_flash_read and spi_flash_write works like a charm but when we mmap the region and then write to the region using spi_flash_write, mmaped region doesn't seem to have updated. 

Provided them pointer, I will be more than happy to work with you to get this resolved. 

Thanks.

igrr, 2022-07-20:
Hi @manimathma, thank you for reporting this bug. The following patch should solve the problem, please give it a try.
[0001-hw-misc-esp32_dport-fix-cache-invalidation.patch.txt](https://github.com/espressif/qemu/files/9150316/0001-hw-misc-esp32_dport-fix-cache-invalidation.patch.txt)

manimathma, 2022-07-20:
@igrr Thanks for the quick fix. It does seem to solve the problem.

igrr, 2022-08-02:
Should be fixed in release https://github.com/espressif/qemu/releases/tag/esp-develop-20220802.

closed 2022-08-02 by igrr