espressif/qemu #49: emulated system seems to boot, but then it just hangs

laurento, 2022-07-19:
Since I can't get my own project running on qemu, I compiled one of the esp-idf examples, the hello_world, but I'm experiencing the very same issue.

Running `qemu-system-xtensa -nographic -M esp32 -m 4 -drive file=build/merged_qemu.bin,if=mtd,format=raw` the system seems to boot as expected but the it simply hangs after listing GPIOs. The code should start printing out messages like _Restarting in %d seconds..._ and eventually reboot the esp32. Any ideas?

I'm using a freshly installed version of esp-idf version 4.4.1 and the latest available release of this project qemu-esp-develop-20220203.

```
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14788
load:0x40080400,len:3792
entry 0x40080694
I (1008) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (1018) boot: compile time 01:18:56
I (1034) boot: chip revision: 0
I (1044) boot.esp32: SPI Speed      : 40MHz
I (1046) boot.esp32: SPI Mode       : DOUT
I (1047) boot.esp32: SPI Flash Size : 4MB
I (1066) boot: Enabling RNG early entropy source...
I (1091) boot: Partition Table:
I (1092) boot: ## Label            Usage          Type ST Offset   Length
I (1095) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (1099) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (1102) boot:  2 factory          factory app      00 00 00010000 00100000
I (1118) boot: End of partition table
I (1134) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e404h (123908) map
I (1358) esp_image: segment 1: paddr=0002e42c vaddr=3ffb0000 size=01bech (  7148) load
I (1406) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=9954ch (628044) map
I (2252) esp_image: segment 3: paddr=000c9574 vaddr=3ffb1bec size=01fc0h (  8128) load
I (2293) esp_image: segment 4: paddr=000cb53c vaddr=40080000 size=15df8h ( 89592) load
I (2449) esp_image: segment 5: paddr=000e133c vaddr=50000000 size=00010h (    16) load
I (2498) boot: Loaded app from partition at offset 0x10000
I (2501) boot: Disabling RNG early entropy source...
I (2533) cpu_start: Pro cpu up.
I (2547) cpu_start: Starting app cpu, entry point is 0x400812bc
I (2069) cpu_start: App cpu up.
I (4726) cpu_start: Pro cpu start user code
I (4734) cpu_start: cpu freq: 160000000
I (4736) cpu_start: Application information:
I (4737) cpu_start: Project name:     hello_world
I (4738) cpu_start: App version:      7084a1b-dirty
I (4739) cpu_start: Compile time:     Jul 15 2022 00:45:22
I (4741) cpu_start: ELF file SHA256:  e669cc31183c9273...
I (4743) cpu_start: ESP-IDF:          v4.4.1-dirty
I (4754) heap_init: Initializing. RAM available for dynamic allocation:
I (4760) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (4763) heap_init: At 3FFC0340 len 0001FCC0 (127 KiB): DRAM
I (4764) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (4765) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (4766) heap_init: At 40095DF8 len 0000A208 (40 KiB): IRAM
I (4908) spi_flash: detected chip: gd
I (4962) spi_flash: flash io: dio
I (5057) cpu_start: Starting scheduler on PRO CPU.
I (20) cpu_start: Starting scheduler on APP CPU.
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 0, 4MB external flash
Minimum free heap size: 236436 bytes
I (8631) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8641) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8651) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8651) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8651) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8651) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8661) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (8661) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
```

igrr, 2022-07-20:
@laurento I have tried building and running the hello_world app from IDF v4.4.1, but couldn't reproduce the issue. The app has booted up successfully and rebooted after printing 10 lines.

Do you have any modifications in the hello_world example? Are you using the default sdkconfig or have you modified anything using menuconfig?

If you could upload an archive with the entire project directory (including the built binaries) it could help in reproducing and tracking down this issue.

igrr, 2022-08-03:
@laurento Do you have any more feedback for this issue?

laurento, 2022-08-03:
Sorry @igrr for the delay, but I'm traveling and I don't have my dev env with me :) If you don't mind, I'd like to make some more tests on a clean env. I should be able to provide some feedback in a few weeks.