espressif/qemu #50: Planning to push to upstream at some point?

cmorganBE, 2022-07-25:
I ask because it looks like this repo has been relatively quiet and was looking for a sign it has long term espressif commitments in terms of support. For our use it would be super valuable to have this as a tool to use for system level pure software tests.

igrr, 2022-07-25:
@cmorganBE we do currently aim to rebase our patches on each upstream release. With each such release we also merge fixes and features which have been accumulated.

With 7.0.0 this got quite a bit delayed, to the point that there's going to be 7.1.0 soon and we still haven't done the 7.0.0 based release yet. The blocking issue was recently resolved though, and we will release a 7.0.0 based version this week.

With regards to upstreaming, currently it is lower priority than adding support for more chips (e.g. ESP32-C3). Adding support more chips will require significant refactoring of the existing code. Once we are more sure about the code structure, we will send an RFC to the upstream project about the possibility of merging our changes.

igrr, 2022-08-03:
Closing, assuming that the question has been answered.

closed 2022-08-03 by igrr