espressif/qemu #51: How to redirect stdio to a socket to get programmatic control via the console?

cmorganBE, 2022-07-25:
I'm trying to get access to stdout via a serial port device so I can programmatically interact with the system, get system test results etc. Maybe this isn't the correct approach but I found online how to map devices:

```
~/projects/qemu/build/qemu-system-xtensa -nographic -machine esp32 -m 2M \
-serial stdio \
-drive 'file=build/flash_image.bin,if=mtd,format=raw'
```

but this results in:

```
QEMU 6.2.0 monitor - type 'help' for more information
(qemu) qemu-system-xtensa: -serial stdio: cannot use stdio by multiple character devices
qemu-system-xtensa: -serial stdio: could not connect serial device to character backend 'stdio'
```

igrr, 2022-07-25:
Hi @cmorganBE, please see if https://github.com/espressif/qemu/wiki#using-esptoolpy-and-espefusepy-to-interact-with-qemu helps.

For test case automation you may consider using qemu together with https://github.com/espressif/pytest-embedded, it includes qemu integration.

cmorganBE, 2022-07-25:
@igrr can I use esptool.py to bind to the console or send read data from it? The wiki doesn't seem to mention that use case.

If I only had the ability to look at output data (maybe I should redirect the qemu stdout to my program since it comes out on the terminal when running qemu?) so I can see if a test passes or fails. I'm fine putting the tests into the image builds themselves.

Oh maybe this is already working? If I run qemu with esp example that has console support can I type into the terminal and it goes to the console like it was a serial character? I'll have to try that example.

I'll look at that example of test automation as we'd like to build up our tests here, thank you for the link.

cmorganBE, 2022-07-25:
Yeah I can type into the terminal and interact with the esp console example so I should be able to redirect qemu output if I start it in a script and look at the test success / failure output. Still have to look at the other links you gave above.

igrr, 2022-07-25:
> If I only had the ability to look at output data (maybe I should redirect the qemu stdout to my program since it comes out on the terminal when running qemu?) so I can see if a test passes or fails.

Yes, you can definitely do that! I suggest spending some time reading Qemu manual, especially the command line flags description (-serial, -monitor, etc). The manual for upstream qemu is applicable for this fork as well.

binary1230, 2022-07-25:
I'm new to qemu specifics as well, but loving this esp32 port. thanks @igrr so much for working on it.

Here's a few commands I was using recently to get stuff like serial IO working. This is assuming you're running on linux:

# ESP32 serial console redirect to /dev/pts/1 on the host. you can use whatever tool you like to talk via /dev/pts/1 at that point.
# example: 
```bash
# start QEMU
qemu-system-xtensa -serial pty  ..............

# while that's running, in another terminal monitor with whatever you like. such as:
# emulated Esp32 with 'idf.py monitor':
idf.py monitor -p /dev/pts/1

# or stuff like:
cat /dev/pts/1 | whatever
cat /dev/pts/1 > somefile
echo "some input" > /dev/pts/1
```

also you can connect serial via TCP.  qemu has a 'telnet' option but I couldn't get it to behave.  raw TCP seems to work great though:
```
# open a socket on port 2424 [arbitrary] to get the output for UART0
qemu-system-xtensa --serial tcp::2424,server,nowait  ......  

# connect to it now with any external program like netcat, putty, etc on port 2424

binary1230, 2022-07-25:
oh and, if you do use 'idf.py monitor' or esptool for serial console, I was only able to get it to work with the ```--no-reset``` option (only bundled in esp-idf 5.0 SDK or later OR you can remove references in serial_reader.py around line 49 to anything that references ```self.serial.dtr```)

[or, just use any other monitoring tool also. i happened to want the address translation for CPU panics]

I've been putting together a prebuilt docker container that is trying to make things a bit easier to dev and debug with qemu+esp32, I have a bunch more to add to it soon.
https://github.com/unit-e/esp-idf-qemu/

you can get full debugging via something like jetbrain's Clion in a docker container running qemu, it's just... quite a ride to get there and some parts are difficult to fully automated.

cmorganBE, 2022-07-28:
@igrr how does one run the tests in https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib?

I've done:

```
cd ~/esp/esp-idf
./install.sh esp32

pip3 install pytest
pip3 install pytest-embedded
pip3 install pytest-embedded-idf
pip3 install pytest-embedded-serial
pip3 install pytest-embedded-serial-esp
apt install python3-dev
pip3 install netifaces
pip3 install pytest-embedded-qemu

idf.py -B build_esp32_default build

pytest --target esp32
```

```
root@86ed30487821:~/esp/esp-idf/examples/build_system/cmake/import_lib# pytest --target esp32
================================================== test session starts ===================================================
platform linux -- Python 3.10.4, pytest-7.1.2, pluggy-1.0.0
rootdir: /root/esp/esp-idf, configfile: pytest.ini
plugins: embedded-0.7.7
collected 1 item                                                                                                         

pytest_import_lib.py::test_pytest_host 
----------------------------------------------------- live log setup -----------------------------------------------------
2022-07-28 01:18:53 INFO find valid binary path: /root/esp/esp-idf/examples/build_system/cmake/import_lib/build_esp32_default
ERROR

========================================================= ERRORS =========================================================
___________________________________________ ERROR at setup of test_pytest_host ___________________________________________
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded/plugin.py:1166: in pytest_fixture_setup
    return self._pytest_fixturedef_exec(fixturedef, request, kwargs)
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded/plugin.py:1122: in _pytest_fixturedef_exec
    result = call_fixture_func(fixturefunc, request, kwargs)
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded/plugin.py:358: in wrapper
    res = func(*args, **kwargs)
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded/plugin.py:973: in serial
    return cls(**_drop_none_kwargs(kwargs))
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded_idf/serial.py:50: in __init__
    super().__init__(
/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded_serial_esp/serial.py:77: in __init__
    raise ValueError('Couldn\'t auto detect chip. Please manually specify with "--port"')
E   ValueError: Couldn't auto detect chip. Please manually specify with "--port"
--------------------------------------------------- Captured log setup ---------------------------------------------------
INFO     root:conftest.py:171 find valid binary path: /root/esp/esp-idf/examples/build_system/cmake/import_lib/build_esp32_default
================================================ short test summary info =================================================
ERROR pytest_import_lib.py::test_pytest_host - ValueError: Couldn't auto detect chip. Please manually specify with "--p...
==================================================== 1 error in 0.71s ====================================================
root@86ed30487821:~/esp/esp-idf/examples/build_system/cmake/import_lib# 
```

Feels like I'm a little close but still missing something.

cmorganBE, 2022-07-28:
And this is all in a ubuntu:latest container btw, so I can recreate and not screw up my local 4.x esp-idf install.

igrr, 2022-08-02:
@cmorganBE Sorry for the late reply. All steps up to the point where you run `pytest` are okay.

Then you need:
```
# add the directory where qemu-system-xtensa is located to PATH
export PATH=/path/to/qemu/bin:$PATH
# run pytest
pytest --embedded-services idf,qemu --target esp32
```

Overriding the build directory using `-B` flag is not required, by the way. If you simply run `idf.py build`, pytest-embedded will also find the binary in the `build` directory.

cmorganBE, 2022-08-03:
@igrr thank you much, let me give that a shot and report back on outcome

cmorganBE, 2022-08-10:
@igrr haven't gotten back to this one just yet but its firmly on the schedule

closed 2023-02-24 by igrr