espressif/qemu #58: Failed to reboot on esp_restart after successful OTA

greenaddress, 2022-11-25:
It seems the latest version of espressif qemu [esp-develop-20220919](https://github.com/espressif/qemu/releases/tag/esp-develop-20220919)  fails to reboot on calling esp_restart after a successful OTA with esp idf [v4.4.3](https://github.com/espressif/esp-idf/releases/tag/v4.4.3)

FYI https://github.com/espressif/qemu/releases/tag/esp-develop-20220802 works with both v4.4.2 and v4.4.3 - i.e. after a successful OTA calling esp_restart restarts the firmware

A workaround that seems to work is to call qemu with -no-reboot when launched and manually start qemu when we expected it to shut down after an esp_restart

Using ESP32 v3 eco mode.

igrr, 2022-11-30:
Thanks for reporting this @greenaddress. We haven't noticed this regression because all our internal tests happen run QEMU with `-no-reboot`. Will take a look which change has caused this.

igrr, 2022-12-27:
@greenaddress Unfortunately I wasn't able to reproduce the issue. I have built the hello_world example from IDF v4.4.3-316-ge86181704a (recent release/v4.4 branch) and it restarts successfully. Do you see the issue happening only after an OTA, or also if simply calling esp_restart from app_main?

closed 2023-02-24 by igrr

greenaddress, 2023-05-26:
for what is worth, using [esp-develop-8.0.0-20230522](https://github.com/espressif/qemu/releases/tag/esp-develop-8.0.0-20230522) with idf 5.0.2 solved the issue.