espressif/qemu #62: Building from tar.gz archive fails due to missing submodules

paulbartell, 2023-01-30:
Attempting to run the configure script with the release tar.gz file fails with the following error:

```
qemu-esp-develop-20220919 % ./configure --target-list=xtensa-softmmu \
    --enable-gcrypt \
    --enable-debug --enable-sanitizers \
    --disable-strip --disable-user \
    --disable-capstone --disable-vnc \
    --disable-sdl --disable-gtk
Using './build' as the directory for build output

ERROR: missing GIT submodules

This is not a GIT checkout but module content appears to
be missing. Do not use 'git archive' or GitHub download links
to acquire QEMU source archives. Non-GIT builds are only
supported with source archives linked from:

  https://www.qemu.org/download/#source

Developers working with GIT can use scripts/archive-source.sh
if they need to create valid source archives.
```

igrr, 2023-02-09:
@paulbartell I think this is, unfortunately, the standard behavior of source archives in Github Releases.
There is nothing we can do in the repository settings to make Github include submodules in the automatically-added .tar.gz and .zip artifacts.

closed 2023-02-24 by igrr

AssassinK786, 2023-05-05:
> @paulbartell I think this is, unfortunately, the standard behavior of source archives in Github Releases. There is nothing we can do in the repository settings to make Github include submodules in the automatically-added .tar.gz and .zip artifacts.

So we have git clone the repo and then build that what you are saying @igrr ?