espressif/qemu #66: Wrong Flash size detection, cannot use flash image more that 4Mb

listout, 2023-02-17:
This issue can be traced back to commit 0152246eb3ebdcb94170c801d93ef20fe682de4a, specifically this [line](https://github.com/espressif/qemu/commit/0152246eb3ebdcb94170c801d93ef20fe682de4a#diff-24bfd73970d5c27614282cc07c19af9ba62bebf6f3063e30d513999210fe8f47R668) where 4MB (or gd25q32) is being hard coded as the flash_dev for some reason. Thus resulting in `E (6755) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.`

Currently the fix is to go into `hw/xtensa/esp32.c` file and edit line 688, replacing `gd25q32` with `gd25q64`.

listout, 2023-02-23:
ping

igrr, 2023-02-23:
Thanks for reporting this issue @listout! If you open a PR with your proposed change, I can merge it into the next release based on QEMU 8.0.

listout, 2023-02-23:
I'll do that