espressif/qemu #73: Issues running binary with version 7.2.0 on Windows

timkoers, 2023-05-07:
Today I've compiled espressifs qemu on Windows using Mingw-64, as my GitLab runner uses the docker image containing 7.2.0 (v4.1.0-28899-g85ff1897fe)

On my Windows machine it's version 7.2.0 (v4.1.0-28895-g1f88fe7c85-dirty.

I'm starting both instances with the exact same script and the Windows version gives me a 
``qemu-system-xtensa: Error: -bios argument not set, and ROM code binary not found (1)`` error whilst the containered version continues to run the binary.

It's the exact same binary that gets run on both, so I don't think that I'm doing anything wrong here.
Version 6.2.0 worked fine!

Full log:
```bash
Adding SPI flash device
qemu-system-xtensa: Error: -bios argument not set, and ROM code binary not found (1)
Program received signal CTRL_BREAK_EVENT (None)
```

igrr, 2023-05-23:
@timkoers Without more information about the steps you use to build and run QEMU on Windows, I'm afraid there's little I can do to help you troubleshoot this issue. It typically happens when the ROM binary file isn't installed along with QEMU binary.

Locating the ROM binary is done using `qemu_find_file`, which is the same that every other machine in QEMU uses:
https://github.com/espressif/qemu/blob/273c5c21928ae63128408e428faf757851a24ecb/hw/xtensa/esp32.c#L843-L846

You can try to use a debugger to step through `qemu_find_file` and check why it isn't finding the ROM binary. You can also try to launch QEMU with some other machine to see if it can find any other binary. Of course, do check that your QEMU installation actually includes the ROM binaries in `<install_path>/share/qemu/` directory.

igrr, 2023-12-12:
Since the release [esp-develop-8.1.3-20231206](https://github.com/espressif/qemu/releases/tag/esp-develop-8.1.3-20231206) we are providing the binaries for Windows, we have checked that they are able to load the ROM code.

closed 2023-12-12 by igrr