espressif/qemu #76: eFuse BLOCK0 MAC

BG7YWL, 2023-06-15:
How to specify the MAC address of eFuse BLOCK0.
The ID I obtained through machine.unique_id() is b'\x00\x00\x00\x00\x00\x00'

igrr, 2023-06-15:
Could you please clarify what `machine.unique_id()` is? It looks like a function call, but in which language/environment/library?

BG7YWL, 2023-06-15:
ESP32-D0WD-V3，MicroPython
[image]

igrr, 2023-06-15:
I see, in this case you need to do the following:

1. Enable eFuse emulation: https://github.com/espressif/qemu/wiki#specifying-efuse-storage
2. Generate eFuse binary, I have put a script here: https://gist.github.com/igrr/3537f8ddb03a50aa980e2e0c121c7581

closed 2023-06-21 by BG7YWL