espressif/qemu #77: ESP32: segfault in qemu_coroutine_switch

redfast00, 2023-06-22:
I built the `hello_world` example from the IDF and tried to run it (as described in the wiki):

```
$ build/qemu-system-xtensa -nographic -machine esp32 -drive file=../hello_world/build/flash_image.bin,if=mtd,format=raw
==50992==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
AddressSanitizer:DEADLYSIGNAL
=================================================================
==50992==ERROR: AddressSanitizer: SEGV on unknown address 0x7f76fffbb420 (pc 0x55c20e020787 bp 0x7f77009bbe40 sp 0x7f77009bbcc0 T2)
==50992==The signal is caused by a READ memory access.
    #0 0x55c20e020787 in qemu_coroutine_switch ../util/coroutine-ucontext.c:315
    #1 0x55c20e01f722 in coroutine_trampoline ../util/coroutine-ucontext.c:178
    #2 0x7f778c1842cf  (/usr/lib/libc.so.6+0x502cf) (BuildId: 2f005a79cd1a8e385972f5a102f16adba414d75e)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV ../util/coroutine-ucontext.c:315 in qemu_coroutine_switch
Thread T2 created by T0 here:
    #0 0x7f778d64a4a7 in __interceptor_pthread_create /usr/src/debug/gcc/gcc/libsanitizer/asan/asan_interceptors.cpp:208
    #1 0x55c20dfe5077 in qemu_thread_create ../util/qemu-thread-posix.c:581
    #2 0x55c20db3a5f0 in mttcg_start_vcpu_thread ../accel/tcg/tcg-accel-ops-mttcg.c:153
    #3 0x55c20d469e53 in qemu_init_vcpu ../softmmu/cpus.c:648
    #4 0x55c20d84deba in xtensa_cpu_realizefn ../target/xtensa/cpu.c:177
    #5 0x55c20db4940f in device_set_realized ../hw/core/qdev.c:510
    #6 0x55c20db5ed3a in property_set_bool ../qom/object.c:2285
    #7 0x55c20db5a995 in object_property_set ../qom/object.c:1420
    #8 0x55c20db63b95 in object_property_set_qobject ../qom/qom-qobject.c:28
    #9 0x55c20db5aedd in object_property_set_bool ../qom/object.c:1489
    #10 0x55c20db483b4 in qdev_realize ../hw/core/qdev.c:292
    #11 0x55c20d6578fc in esp32_soc_realize ../hw/xtensa/esp32.c:323
    #12 0x55c20db4940f in device_set_realized ../hw/core/qdev.c:510
    #13 0x55c20db5ed3a in property_set_bool ../qom/object.c:2285
    #14 0x55c20db5a995 in object_property_set ../qom/object.c:1420
    #15 0x55c20db63b95 in object_property_set_qobject ../qom/qom-qobject.c:28
    #16 0x55c20db5aedd in object_property_set_bool ../qom/object.c:1489
    #17 0x55c20db483b4 in qdev_realize ../hw/core/qdev.c:292
    #18 0x55c20d65af4d in esp32_machine_init ../hw/xtensa/esp32.c:782
    #19 0x55c20d0a5387 in machine_run_board_init ../hw/core/machine.c:1409
    #20 0x55c20d489575 in qemu_init_board ../softmmu/vl.c:2513
    #21 0x55c20d4899f8 in qmp_x_exit_preconfig ../softmmu/vl.c:2609
    #22 0x55c20d48e6d0 in qemu_init ../softmmu/vl.c:3612
    #23 0x55c20db40274 in main ../softmmu/main.c:47
    #24 0x7f778c15784f  (/usr/lib/libc.so.6+0x2384f) (BuildId: 2f005a79cd1a8e385972f5a102f16adba414d75e)

==50992==ABORTING
```

The output of esptool to build the flash_image.bin is:

```
$ esptool.py --chip esp32 merge_bin --fill-flash-size 4MB -o flash_image.bin @flash_args
esptool esp32 merge_bin --fill-flash-size 4MB -o flash_image.bin --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Wrote 0x400000 bytes to file flash_image.bin, ready to flash to offset 0x0
```

redfast00, 2023-06-22:
Note that on Arch Linux, we have a really new version of xkeyboard-config, so you will need to apply this one-line patch to get it to build  [[PATCH] pc-bios/keymaps: Use the official xkb name for Arabic layout, not the legacy synonym](https://patchew.org/QEMU/20230620162024.1132013-1-peter.maydell@linaro.org/)

igrr, 2023-06-28:
Given

> ==50992==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!

and that the crash happens in

>     #0 0x55c20e020787 in qemu_coroutine_switch ../util/coroutine-ucontext.c:315

could you try to build QEMU without the address sanitizer enabled?

redfast00, 2023-06-28:
Oh, that's really obvious, my bad. Passing `--disable-sanitizers` to `./configure` indeed fix it, thank you :)

(Note that in the meantime, I built this in a debian docker, which also printed `==7==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!`, but did not crash)

closed 2023-06-28 by redfast00