espressif/qemu #79: UART Echo Example not working as expected in ESP32C3 Qemu

lcgamboa, 2023-06-30:
The [idf UART Echo Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo) not working as expected in ESP32C3 Qemu. 
In the figure below on the left is the output on a real C3 chip and on the right the output of Qemu's ESP32C3 for keys typed in sequence  ( 1, 2 , 3 ...).

[image]

The example works correctly with Qemu for ESP32.

o-marshmallow, 2023-07-03:
Hi @lcgamboa ,

Thanks for reporting this issue. As far as I know, this example requires an external USB-to-UART adapter as it uses another UART controller than the default one, with given pins from `Kconfig`.

I cannot reproduce what you are getting, even on the real chip. What is your configuration for this example?

lcgamboa, 2023-07-03:
I'm sorry I forgot to let you know that I modified the code to use the default pins for ease of use with many different processors. I modified the lines [29 and 30](https://github.com/espressif/esp-idf/blob/cf7e743a9b2e5fd2520be4ad047c8584188d54da/examples/peripherals/uart/uart_echo/main/uart_echo_example_main.c#L29) of file uart_echo_example_main.c from :
```
#define ECHO_TEST_TXD (CONFIG_EXAMPLE_UART_TXD)
#define ECHO_TEST_RXD (CONFIG_EXAMPLE_UART_RXD)
```
to:
```
#define ECHO_TEST_TXD (UART_PIN_NO_CHANGE)
#define ECHO_TEST_RXD (UART_PIN_NO_CHANGE)
```

o-marshmallow, 2023-07-06:
Hi @lcgamboa ,

Thanks for the details, I also needed to use the UART port number 0 in the `menuconfig` to reproduce the issue.
I will investigate this issue.

closed 2023-08-03 by espressif-bot

lcgamboa, 2023-08-05:
Thanks for the support, now the example works correctly.

lcgamboa, 2023-08-06:
I did some more tests and it seems that there is some difference when the sent string is greater than 128 bytes. For real hardware the string limit is 1024 bytes, for Qemu esp32-c3 the limit is a multiple of 128. In the figure below a string of 2048 bytes is sent, for real hardware (top terminal) it is split into two parts of 1024 (in the most times). For Qemu esp32-c3 (bottom terminal) it is always split into several parts with sizes x*128 randomly. Same with ESP32 for hardware and on Qemu esp32. Apparently this is not a problem and is probably due to a limitation in the Qemu implementation.

[image]

o-marshmallow, 2023-08-08:
Hi @lcgamboa ,

The reason behind this is that the ESP32-C3 UART controller underneath uses the ESP32 UART implementation, which explains why it has the same behavior. This choice was made to quickly get something up and running while developping other controllers.
If you think this is a critical bug and it needs to be fixed as soon as possible, feel free to open a new issue

lcgamboa, 2023-08-09:
Hi @o-marshmallow ,

I believe it is not critical, this limitation is because the buffers always have a[ fixed size of 128](https://github.com/espressif/qemu/blob/f584a02be92b8db8d10acb6eac98498b899b0898/include/hw/char/esp32_uart.h#L9) in the Qemu implementation.

Another problem that occurs is that when the uart receive threshold is set to a value lower than 128, the[ interruption](https://github.com/espressif/qemu/blob/f584a02be92b8db8d10acb6eac98498b899b0898/hw/char/esp32_uart.c#L43C5-L43C75) only happens when the buffer reaches 128 and [`throttle_rx=true`](https://github.com/espressif/qemu/blob/f584a02be92b8db8d10acb6eac98498b899b0898/hw/char/esp32_uart.c#L280C12-L280C23) occurs. It's not critical either, but it can disturb an application that needs to process exactly the configured threshold.