espressif/qemu #89: Crash when executing QEMU

Joebeazelman, 2023-12-16:
I tried to follow the instruction for QEMU RGB Panel and I successfully built the project and image. When I tried to execute it using QEMU, a window flashed briefly on the screen followed by an crash report on MacOS:

I typed the following command according to the tutorial:
>  qemu-system-riscv32 \
>     -icount 3 \
>     -machine esp32c3 \
>     -drive file=flash_image.bin,if=mtd,format=raw \
>     -display sdl \
>     -serial stdio

After the crash report was displayed, the following was displayed on the console:
```
Adding SPI flash device
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e34
entry 0x403cc71a
I (0) boot: ESP-IDF v5.3-dev-892-g692c1fcc52-dirty 2nd stage bootloader
I (0) boot: compile time Dec 14 2023 19:29:29
I (0) boot: chip revision: v0.3
I (0) boot.esp32c3: SPI Speed      : 80MHz
I (0) boot.esp32c3: SPI Mode       : SLOW READ
I (0) boot.esp32c3: SPI Flash Size : 2MB
I (0) boot: Enabling RNG early entropy source...
I (1) boot: Partition Table:
I (1) boot: ## Label            Usage          Type ST Offset   Length
I (1) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (1) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (1) boot:  2 factory          factory app      00 00 00010000 00100000
I (1) boot: End of partition table
I (1) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=0eb08h ( 60168) map
I (7) esp_image: segment 1: paddr=0001eb30 vaddr=3fc8b000 size=01328h (  4904) load
I (8) esp_image: segment 2: paddr=0001fe60 vaddr=40380000 size=001b8h (   440) load
I (8) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=46980h (289152) map
I (35) esp_image: segment 4: paddr=000669a8 vaddr=403801b8 size=0acc0h ( 44224) load
I (44) boot: Loaded app from partition at offset 0x10000
I (44) boot: Disabling RNG early entropy source...
I (44) cpu_start: Unicore app
I (53) cpu_start: Pro cpu start user code
I (53) cpu_start: cpu freq: 160000000 Hz
I (53) cpu_start: Application information:
I (53) cpu_start: Project name:     lcd_qemu_rgb_panel
I (53) cpu_start: App version:      1
I (53) cpu_start: Compile time:     Dec 14 2023 19:29:13
I (53) cpu_start: ELF file SHA256:  677c1292d...
I (53) cpu_start: ESP-IDF:          v5.3-dev-892-g692c1fcc52-dirty
I (53) cpu_start: Min chip rev:     v0.3
I (54) cpu_start: Max chip rev:     v1.99 
I (54) cpu_start: Chip rev:         v0.3
I (54) heap_init: Initializing. RAM available for dynamic allocation:
I (54) heap_init: At 3FC957A0 len 0002A860 (170 KiB): RAM
I (54) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (54) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (54) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (54) spi_flash: detected chip: gd
I (54) spi_flash: flash io: dio
W (54) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (55) sleep: Configure to isolate all GPIO pins in sleep state
I (55) sleep: Enable automatic switching of GPIO sleep configuration
I (55) main_task: Started on CPU0
I (55) main_task: Calling app_main()
I (55) example: Install RGB LCD panel driver
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow geometry should only be modified on the main thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff811924fa6 __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff81141a231 objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff811948dbc -[NSException raise] + 9
	3   AppKit                              0x00007ff814ee65f2 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 235
	4   AppKit                              0x00007ff814ef64c5 -[NSView setFrameOrigin:] + 830
	5   AppKit                              0x00007ff815c59ca4 -[NSTitlebarContainerView setFrameOrigin:] + 136
	6   AppKit                              0x00007ff814f00474 -[NSView setFrame:] + 317
	7   AppKit                              0x00007ff814ef3e00 -[NSThemeFrame _updateTitlebarContainerViewFrameIfNecessary] + 406
	8   Alibc++abi: terminating due to uncaught exception of type NSException
[1]    5900 abort      qemu-system-riscv32 -icount 3 -machine esp32c3 -drive  -display sdl -serial 

```

igrr, 2023-12-16:
This issue has been fixed in https://github.com/espressif/qemu/commit/53e8f0a9f7dcc4c2f9f738f01cfd978066b89165.

Joebeazelman, 2023-12-19:
This fix wasn't pushed to the latest official SDK tools. I had to download the latest artifact and replaced my Qemu binaries with them. 

https://github.com/espressif/qemu/actions/runs/7178958612

I'm still getting the same error.

igrr, 2023-12-19:
That's interesting... are you sure you are running the new QEMU binary you have downloaded and not the one installed by the SDK? To check, could you please try running `qemu-system-riscv32 --version` and see what the output is?

Joebeazelman, 2023-12-19:
> That's interesting... are you sure you are running the new QEMU binary you have downloaded and not the one installed by the SDK? To check, could you please try running `qemu-system-riscv32 --version` and see what the output is?

QEMU emulator version 8.1.3 (g53e8f0a9)
Copyright (c) 2003-2023 Fabrice Bellard and the QEMU Project developers

Joebeazelman, 2023-12-19:
Just for your information, I discovered the undocumented **idf.py qemu** command. It logs to the terminal without displaying the virtual LCD windows. I copied its command-line put and removed the  **-nographic** flag which opens up the virtual LCD and crashes. 

Is it possible the code sample program may be the culprit? Could it be it wasn't intended to display any graphics?

igrr, 2023-12-19:
The command is in fact documented [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/tools/qemu.html), however the merge request for graphics support in that command (`idf.py qemu --graphics` or `idf.py qemu -g`) has been merged just a couple days ago, and this change is not yet in the public esp-idf repository on Github. It's a pretty new feature, so apologies for not having every part of it in place yet!

I am running the same example program but unfortunately can't reproduce the issue you are seeing. I am running macOS 13.5.1, which version do you have?

Could you please try the same example, but for the esp32 target? Use `idf.py set-target esp32` to switch the target on the IDF side, and then use qemu-system-xtensa instead of qemu-system-riscv32 when launching QEMU.

If that doesn't help, the only other idea I have is to try building QEMU with `cocoa` graphical backend support, and then launching it with `-display cocoa` instead of `-display sdl`.

Joebeazelman, 2023-12-19:
I'm running sonoma version 14. Let me try the esp32.

Joebeazelman, 2023-12-21:
I was able to successfully run the QEMU emulation with graphics. There are two issues. First, idf detects the qemu binaries as not matching the current release and doesn't add them to the path. Second, while running QEMU, I haven't been able to quit the emulator without terminating the terminal. It doesn't respond to any of the keystrokes.  It also causes side-effects such as stuck menus and disappearing cursors.

igrr, 2023-12-21:
> First, idf detects the qemu binaries as not matching the current release and doesn't add them to the path.

I see, I didn't realize you were overwriting the version downloaded by IDF with a different one. Yeah, that won't work. Typically when testing a different build of QEMU I have it in a different directory, and simply prepend this directory to PATH in the terminal where I am testing it. You can run `which -a qemu-system-riscv32` and verify that the version you need is the first one on the list.

> Second, while running QEMU, I haven't been able to quit the emulator without terminating the terminal

If you are running `idf.py qemu` then you get into a regular QEMU session, where you can switch between the serial output and QEMU monitor (i.e. "command interface") using Ctrl+A C. Then you can press x for exit and press Enter. You can find more details in QEMU manual.

Alternatively, you can run `idf.py qemu monitor` and get the similar experience as `idf.py flash monitor `. You can exit that using Ctrl+] as usual.