espressif/qemu #91: ESP32C3 Qemu w64-mingw32 freeze with Arduino blink

lcgamboa, 2023-12-30:
I'm trying to run codes compiled in the Arduino framework on the Qemu ESP32C3. To work in Qemu it is necessary to add a return value for [saradc interrupts](https://github.com/lcgamboa/qemu/blob/338ca08583f0978e2cca141e56d2706cba1db247/hw/riscv/esp32c3.c#L147). Once this is done, the codes work without problems on Qemu Linux with the command line:
```
qemu-system-riscv32 -M esp32c3 -drive file=blink.bin,if=mtd,format=raw  \
-serial stdio \
-icount shift=3 \  
-global driver=timer.esp32c3.timg,property=wdt_disable,value=true
```

But for the w64-mingw32  version the code always hangs after a few seconds, using gdb it is possible to verify that the code is stopped in the `wfi` instruction. This happens using the Arduino IDE or platformio with the Arduino framework (which uses a precompiled version without debug symbols). Compiling the code using Arduino as an IDF component (with or without debug symbols) the problem does not occur  which makes it difficult to discover its origin.

Without disabling WDT, Qemu is reset several times without freezing.

Any help to identifying the source of the problem would be very helpful.

Test code:
```
#define LED_BUILTIN 5

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Led on pin: ");
  Serial.println(LED_BUILTIN);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   
  Serial.print(millis());
  Serial.println(" LED ON");
  delay(500);                       
  digitalWrite(LED_BUILTIN, LOW);                       
  Serial.print(millis());
  Serial.println(" LED OFF");
  delay(500);  
}
```
The .bin file that not work with  w64-mingw32  saradc int patched version:
[blink.zip](https://github.com/espressif/qemu/files/13799135/blink.zip)

igrr, 2023-12-31:
Thanks for the report. We have just fixed a similar sounding issue, the fix should be part of the next release.

lcgamboa, 2023-12-31:
Thanks for the support. I'll be waiting for the next release.

igrr, 2024-01-09:
@lcgamboa The last 2-3 commits on the esp-develop branch could fix your issue, could you please give it a try?

lcgamboa, 2024-01-09:
Hi @igrr  , when I opened the issue I had already tested it with the modifications from these last three commits.
With these three last commits, Qemu ESP32C3 works better without  the `icont` option than previously on Linux, on Windows it continues to freeze after a few seconds.