espressif/qemu #92: Attach gdb or s argument shows duplicate yank instance

coozoo, 2024-01-24:
I'm trying to run image using qemu with attached gdb but looks like gdb flag doesn't work

```
./qemu-system-xtensa -M esp32 -s -d guest_errors,page -nographic -drive file=flash_4M.bin,if=mtd,format=raw -global driver=timer.esp32.timg,property=wdt_disable,value=true -no-reboot -gdb tcp::1234 -S
```

with tcp port defined I'm getting such error

```
Adding SPI flash device
qemu-system-xtensa: -gdb tcp::1234: duplicate yank instance
```

with shortcut -s

```
Adding SPI flash device
qemu-system-xtensa: -s: duplicate yank instance
```

with attempt to use socket  -gdb chardev:gdb0 it's simply exiting with error code 1. I'm not sure what does it mean

```
Adding SPI flash device
$ echo $?
1
```

When I'm launching without -s -S flag 
I see output with some errors but actually I see output from program

```
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
M25P80: Read id (command 0x90/0xAB) is not supported by device
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
M25P80: Unknown cmd 5a
Invalid read at addr 0x10200C, size 4, region 'er', reason: rejected
Invalid read at addr 0x10200C, size 4, region 'er', reason: rejected
E (1985) esp_core_dump_flash: No core dump partition found!
E (1987) esp_core_dump_flash: No core dump partition found!

Oled screen initialised
cc2500 failed
cc2500 not found

```
Am  I doing something wrong?

How can I attach gdb?

igrr, 2024-01-24:
> -s ... -gdb tcp::1234 -S

According to https://www.qemu.org/docs/master/system/invocation.html, `-s` is a shorthand for `-gdb tcp::1234`, so you are essentially opening the port 1234 twice. Try removing either `-s` or `-gdb tcp::1234`.

coozoo, 2024-01-24:
oh damn yes!  how can I be so stupidly blind

thank you..

closed 2024-01-24 by igrr