espressif/qemu #93: build fail on HEAD

FransM, 2024-02-20:
While trying to build head I get a linker error

```
[1/118] Generating subprojects/dtc/version_gen.h with a custom command
[2/118] Generating qemu-version.h with a custom command (wrapped by meson to capture output)
[3/5] Linking target qemu-system-aarch64
FAILED: qemu-system-aarch64 
cc -m64 -mcx16 @qemu-system-aarch64.rsp
/usr/bin/ld: libcommon.fa.p/hw_nvram_xlnx-efuse.c.o: in function `xlnx_efuse_k256_check':
/home/frans/git/qemu/build/../hw/nvram/xlnx-efuse.c:174: undefined reference to `xlnx_efuse_calc_crc'
/usr/bin/ld: libcommon.fa.p/hw_nvram_xlnx-bbram.c.o: in function `bbram_aes_crc_postw':
/home/frans/git/qemu/build/../hw/nvram/xlnx-bbram.c:249: undefined reference to `xlnx_efuse_calc_crc'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [Makefile:162: run-ninja] Error 1
```

xlnx_efuse_calc_crc is defined in ../hw/nvram/xlnx-efuse-crc.c but that file is not compiled

I'm not familiar enough with the build structure and ninja to come up with a patch.

Build system is Ubuntu 22.04

igrr, 2024-02-20:
You seem to be compiling the `qemu-system-aarch64` target. Is that the emulation target you need, given that you are using the Espressif fork?

FransM, 2024-02-20:
> You seem to be compiling the `qemu-system-aarch64` target. Is that the emulation target you need, given that you are using the Espressif fork?

No, I want to build the esp32 version.
Guess I am confused by the instructions.
I checked out https://github.com/espressif/qemu

I'm at this commit.
commit fe80b1870651ef1227bffd7d4151aa2ce4bcf65f (HEAD -> esp-develop, origin/esp-develop, origin/HEAD)
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Tue Jan 23 17:10:09 2024 +0100

    hw/sd: dwc_sdmmc: save clksrc register as well

The documentation on https://github.com/espressif/qemu says

```
Building

QEMU is multi-platform software intended to be buildable on all modern Linux platforms, OS-X, Win32 (via the Mingw64 toolchain) and a variety of other UNIX targets. The simple steps to build QEMU are:

mkdir build
cd build
../configure
make
```

So I figured that, since I was at the esp fork everything was configured for esp
Guess the README.rst file could be improved.

No idea what I should do to get this building

igrr, 2024-02-20:
Sorry for the confusion — we haven't made any changes to the readme file compared to the upstream version.

Please check the instructions over here: https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/README.md (the link is in the description of this GitHub repo).

FransM, 2024-02-21:
@igrr Thanks for the pointer.
Probably change the README.md file in https://github.com/espressif/qemu or at least add a note in the beginning pointing to the right location. Currently if you go to the archive and just start reading you end up with the above question.
With the page you linked I was able to build qemu.
Leaving it to you to close this or keep it open until the main README.md is updated.

igrr, 2025-02-28:
The issue was related to #100 — we had accidentally lost the build script line related to xlnx-efuse-crc. This is fixed in [esp-develop-9.2.2-20250228](https://github.com/espressif/qemu/releases/tag/esp-develop-9.2.2-20250228) release.

closed 2025-02-28 by igrr