espressif/qemu #94: binary install on ubuntu does not work.

FransM, 2024-02-20:


igrr, 2024-02-20:
It doesn't look like a QEMU-specific issue, you'll probably get the same error for other idf.py commands if the required package is not installed.

Please try installing libslirp and libsdl2.0, and then run install.sh and `. ./export.sh` again, perhaps that is the issue you have encountered: https://github.com/espressif/esp-idf/issues/13152

closed 2024-02-21 by FransM

igrr, 2024-02-21:
I see, that's interesting! I still don't understand why only one idf.py command was failing, as the python environment check is also done for the rest of the commands. But since you have solved the issue, let's keep it closed. We can investigate if it gets reported again.