espressif/qemu #98: ESP32-S3 documentation

PatrickV82, 2024-06-19:
I am trying to use the ESP32-S3 implementation that was released with the version 9.0.0

I have the following questions: 

1. Is there a updated documentation for the bootstrapping pins values?
2.  Does the esp32s3 implementation already support the efuses?
3.  Does the esp32s3 implementation support Flash Encryption and SecureBoot ?

When i try to use the efuse in the same manor as the ESP32-C3 i get the following error 

qemu-system-xtensa -nographic -machine esp32s3 -global driver=esp32s3.gpio,property=strap_mode,value=0x08 -drive file=build/qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32s3.efuse,property=drive,value=efuse -serial tcp::5555,server,nowait

QEMU 9.0.0 monitor - type 'help' for more information
Not initializing SPI Flash
c:\qemu\bin\qemu-system-xtensa: warning: global nvram.esp32s3.efuse.drive has invalid class name

igrr, 2024-06-19:
Hi @PatrickV82,

Currently there is a stub of eFuse implementation for ESP32-S3, which is simply a copy of the one from ESP32-C3. I think if you replace `driver=nvram.esp32s3.efuse` with `driver=nvram.esp32c3.efuse` then QEMU will run. However, I am not sure yet how different ESP32-C3 and ESP32-S3 eFuse modules are, so this might not actually work with ESP-IDF or esptool.

We'll update this issue once ESP32-S3 eFuse support is done!

greenaddress, 2024-06-27:
@igrr is there support for openeth driver in qemu for esp32s3? I've tried to build but had some issue around openeth not building for esp32s3 target (but ok for esp32)

igrr, 2024-06-27:
@greenaddress It should have been added in https://github.com/espressif/esp-idf/commit/f95caa75. Have you enabled [CONFIG_ETH_USE_OPENETH](https://github.com/espressif/esp-idf/blob/f95caa751b6842d54b68e282a08f6dd53012feda/components/esp_eth/Kconfig#L161) in menuconfig?

greenaddress, 2024-07-09:
@igrr We are on 5.2.2 right now, hope 5.2.3 or 5.3.x will have it

igrr, 2024-11-14:
@greenaddress Forgot to update you, openeth driver support for ESP32-S3 has been released in IDF v5.2.3 (https://github.com/espressif/esp-idf/commit/c143e68c9f3d896f93e8de7febda23874103ff24)

igrr, 2024-11-27:
Documentation has been added in https://github.com/espressif/esp-toolchain-docs/commit/ca4ba81a81b97da539a3e7244439e3dda0d51dce.

closed 2024-11-27 by igrr