espressif/qemu #101: canbus support

maedoc, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Support the canbus object per https://www.qemu.org/docs/master/system/devices/can.html

### Use cases

I would like to debug my esp32 app using the twai (or similar) with a virtual can interface on linux

### Alternatives

The linux target, but it doesn't compile with my app.

igrr, 2024-11-11:
@maedoc I've looked into the possibility of CAN support, and it seems very doable. QEMU already has an implementation of SJA1000 controller, we just need to implement a small wrapper device to attach SJA1000 as an MMIO peripheral (similar to hw/net/can/can_pcm3680_pci.c, just without PCI...)

This will only work on Linux hosts, though, since the only CAN host in QEMU is `socketcan`, which is only supported on Linux. I would just like to confirm whether you are okay with using Linux as host OS?

maedoc, 2024-11-12:
> okay with using Linux as host OS? 

yes only been using esp idf on Linux, or Docker on macOS, so that would be fine.

thanks for considering implementing!

wodz, 2025-01-22:
Here you can find patches which implement CAN (TWAI) esp32 peripheral. (I had to change extension to .txt because of some stupid github attachements policy).

1) In order to share canbus with host you need to have real socket can interface or virtual one up. To setup virtual can interface run on the host:
modprobe vcan
ip link add dev vcan0 type vcan
ip link set vcan0 mtu 16
ip link set up vcan0

2) Add params when running patched qemu-system-xtensa:
-object can-bus,id=canbus0 -object can-host-socketcan,id=canhost0,if=vcan0,canbus=canbus0 -global driver=esp32.twai,property=canbus,value=canbus0

Enjoy.

[0001-Implement-esp32-TWAI.txt](https://github.com/user-attachments/files/18506213/0001-Implement-esp32-TWAI.txt)

[0003-cherry-pick-25145a7d77-from-upstream.txt](https://github.com/user-attachments/files/18506221/0003-cherry-pick-25145a7d77-from-upstream.txt)
[0004-SJA1000-fix-RMC-read.txt](https://github.com/user-attachments/files/18506222/0004-SJA1000-fix-RMC-read.txt)

closed 2025-08-19 by Indastri

maedoc, 2025-10-31:
sorry to bump this, but two questions: 

(a) will this get merged in the qemu binaries we get in esp-idf by default?

(b) are the patches only for xtensa or riscv also?

thanks again for the impl effort