espressif/qemu #103: ESP32-S3 in QEMU: PSRAM cannot work for esp32s3 device

dongsheng28849455, 2024-09-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

When I try to enable the PSRAM, I met several issues:
1, Similar with #57, neither SPIRAM_SPEED_80M nore SPIRAM_SPEED_40M can work
2, The ESP32S3MMUEntry e value will be overwritten by [e.reserved](https://github.com/espressif/qemu/blob/cacb3b1e51a0edcd523b0594526c7baff3ffc251/hw/misc/esp32s3_cache.c#L67)
4, dcache was created as [rom device](https://github.com/espressif/qemu/blob/cacb3b1e51a0edcd523b0594526c7baff3ffc251/hw/misc/esp32s3_cache.c#L309), so dbus address cannot be written
3, SSS_PSRAM does not attach to SPI1 for S3 machine
4, RAM size larger than 4 MB not supported, seems S3 can support 32M as maximum.

Is it possible to support PSRAM for S3 also?

### Use cases

In nuttx nsh, we expect the umem can be 32M size.
NuttShell (NSH) NuttX-12.6.0-RC1
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Kmem:     359196      16612     342584      17000     342584     43      1
      Umem:   33161212      10596   33150616      10984   33150616      9      1

igrr, 2024-09-19:
Yes, as you found, PSRAM support is not implemented for S3 yet.

We'll update this issue when that gets done.

dongsheng28849455, 2024-09-20:
> Yes, as you found, PSRAM support is not implemented for S3 yet.
> 
> We'll update this issue when that gets done.

Can I ask will you implement it recently?

dongsheng28849455, 2024-11-08:
I try to fix it by #104, please help review it, thanks.

o-marshmallow, 2024-11-08:
Hello @dongsheng28849455 ,

Sorry for not giving more updates about this. A few days ago, we opened an internal MR to add support for the PSRAM on the S3. It is currently under review.

igrr, 2024-11-11:
PSRAM support for ESP32-S3 has been merged in https://github.com/espressif/qemu/commit/39534c3193c88417b1cc9836169f3433dfff42a5 and https://github.com/espressif/qemu/commit/89fe574c628124d0c2329ba8f05f892e685100df, will appear in the next release.

dongsheng28849455, 2024-11-12:
Thanks for your works, when i test it, seems the [get_eid_by_size](https://github.com/espressif/qemu/blob/1d55ccda6b2dbf643c2e39e6d14eed9926c99000/hw/misc/ssi_psram.c#L24) does not handle the case of [size_mbytes](https://github.com/espressif/qemu/blob/1d55ccda6b2dbf643c2e39e6d14eed9926c99000/hw/misc/ssi_psram.c#L88C28-L88C39) equal to 16M or 32M?

igrr, 2024-11-14:
@dongsheng28849455 That's right. There are no commercially available 16MB or 32MB QSPI PSRAM chips yet, so neither IDF nor QEMU have defined an ID for them.

We have a work-in-progress MR for IDF and for QEMU which defines a couple of fake IDs, which don't correspond to any real PSRAM models. I'll ping you once these MRs are merged, then you will be able to copy the implementation into NuttX.

dongsheng28849455, 2024-11-21:
hi, @igrr , Another question, for the psram spi data transfer, how can we set the dummy size except the time device during[ init](https://github.com/espressif/qemu/blob/a4dfc0864e0d19bd74371eaf7cb25f2c68f2aed8/hw/misc/ssi_psram.c#L106)? I met some case that demand the dummy being 0 or 1. can it be setting dynamically? or any parameter we can set when loading the image?

dongsheng28849455, 2024-11-25:
sorry, here I mentioned is for esp32, just by adding and reading logs. seems the "dummy" must be set as 0 rather than 1 before it can work.

igrr, 2024-11-25:
You are testing this on NuttX, right? Could this be some difference between IDF's PSRAM driver and NuttX driver?

cc @tmedicci

dongsheng28849455, 2024-11-26:
Yes, I'm testing nuttx, but a private one from apache. cc @donghengqaz, BTW, in current impl, seems no bytes in dummy phase insterted for esp32? (in cmd, addr, data sequence) refer to [esp32_spi_transaction](https://github.com/espressif/qemu/blob/a4dfc0864e0d19bd74371eaf7cb25f2c68f2aed8/hw/ssi/esp32_spi.c#L177), thus the "dummy" in sss_psram must be 0.

but have dummy for esp32s3 (in cmd, addr, dummy, data) ([esp32_spi_transaction](https://github.com/espressif/qemu/blob/a4dfc0864e0d19bd74371eaf7cb25f2c68f2aed8/hw/ssi/esp32s3_spi.c#L182))

closed 2025-02-28 by igrr