espressif/qemu #106: latest tag can't clone recursively as repo https://github.com/Zeex/subhook.git  is no longer available

greenaddress, 2024-11-25:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Expected all tag dependencies to be available and allow to complete a full build

### Actual behavior (suspected bug)

git clone fails at submodule https://github.com/Zeex/subhook.git as it is no longer

### Steps to reproduce the behavior

git clone this repo, checkout tag, init/update all submodules recursively

### Project release version

esp-develop-9.0.0-20240606

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

debian

### Shell

ZSH

greenaddress, 2024-11-25:
it seems esp-develop is also affected

igrr, 2024-11-25:
@greenaddress Thank you for the report. We will pick up upstream fix https://github.com/qemu/qemu/commit/b1e35c760acd9a85a4469c69465815ec2e2e6eba in the next release.

For now, please update your build process to not do recursive cloning of submodules, please see https://gitlab.com/qemu-project/qemu/-/issues/2660#note_2201708060 and https://gitlab.com/qemu-project/qemu-web/-/merge_requests/10.

greenaddress, 2024-11-25:
@igrr thanks!

igrr, 2025-02-28:
Should be fixed in [esp-develop-9.2.2-20250228](https://github.com/espressif/qemu/releases/tag/esp-develop-9.2.2-20250228) release.

closed 2025-02-28 by igrr