espressif/qemu #107: Unable to set Ethernet mac address from qemu command line

mzakharocsc, 2024-11-27:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

when specifying custom mac on qemu command line like this: ```-nic user,model=open_eth,mac=00:00:00:00:00:05```  , the value does not reflect expected bytes when read using the following api:
```esp_read_mac(mac_addr, ESP_MAC_ETH)```

### Actual behavior (suspected bug)

mac_addr always shows up as "00:00:00:00:00:03", regardless of what qemu's parameter  'mac=XX:XX;XX:XX:XX:XX' is set to.

### Steps to reproduce the behavior

idf.py qemu monitor 

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Shell

Bash

mzakharocsc, 2024-11-27:
Running examples/ethernet/basic, which does       `esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);` returns '00:00:00:00:00:03' also, thats because I suspect in `static esp_err_t emac_opencores_init(esp_eth_mac_t *mac`  in esp_eth_mac_openeth.c it also uses `esp_read_mac()` to set mac address of the interface.  

Please confirm

igrr, 2024-11-27:
Ah yes, you are totally right, I have missed that part.

I see two things we could do here:
1. Modify emac_opencores_init, adding more complex logic around the selection of the MAC address:
   - Get the base MAC address. If it is non-zero, then use `esp_read_mac` to obtain the address derived for the Ethernet interface. This would handle the case when the emulated eFuses do contain a valid MAC address.
   - If the base MAC address is zero, try to get the address from the (emulated) ethernet MAC. If non-zero, use that address. This would allow setting the address via `-nic user,model=open_eth,mac=....` argument.
   - If both are zero, fall back to some default address.
2. Add `--mac-addr` option in `idf.py qemu` wrapper to allow setting the base address in eFuses.

Does that sound reasonable to you?

mzakharocsc, 2024-11-27:
1. is the preferred option in my opinion

I tried to read the MAC address like so:
```C
    // Initialize the MAC
    openeth_reset();
    openeth_set_tx_desc_cnt(TX_BUF_COUNT);

    uint32_t mac0 = REG_READ(OPENETH_MAC_ADDR0_REG);
    uint32_t mac1 = REG_READ(OPENETH_MAC_ADDR1_REG);

    printf("MAC0=%#lx  MAC1=%#lx\n", mac0, mac1);
 ```
inside [esp_eth_mac_openeth.c](https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/esp_eth/src/openeth/esp_eth_mac_openeth.c#L291), however I am getting all 0's even when  `-nic user,model=open_eth,mac=00:00:00:00:00:05` is set on QEMU parameters

Please advise.

igrr, 2024-11-27:
Looks like openeth emulation doesn't call `qemu_macaddr_default_if_unset` to initialize the MAC address, like other ethernet adapter emulations in QEMU do, for example

https://github.com/espressif/qemu/blob/d2f893f982067986ef444998a824c853592a1d95/hw/net/ne2000-pci.c#L69

Will fix that as well.

igrr, 2024-12-11:
QEMU part of the fix merged in https://github.com/espressif/qemu/commit/b726cfe451fbc033d94856feaec067620cc151a0, IDF part of the fix is in review and will be linked here once merged.

igrr, 2025-02-28:
QEMU fix has been released in [esp-develop-9.2.2-20250228](https://github.com/espressif/qemu/releases/tag/esp-develop-9.2.2-20250228).

closed 2025-02-28 by igrr