espressif/qemu #108: [Documentation/Feature] Remote output emulation

mtribiere, 2024-12-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Hello, 
I was wondering if the connecting an external process to qemu to inspect and inject GPIO states was possible. 
Maybe the feature is already implemented, but I am unable to find any substantial amount of documentation to help me understand this process. 

Thanks

### Use cases

The idea for me would be to build an emulation/testing software to avoid flashing the real hardware again and again.
Here is an example of a similar software (that doesn't use qemu):
[https://github.com/Open-Smartwatch/open-smartwatch-os?tab=readme-ov-file#osw-emulator](Here)

igrr, 2024-12-05:
Hi @mtribiere, there was a similar request in https://github.com/espressif/qemu/issues/41 but so far I haven't found a reasonable way to implement this.

If you need to observe and inject GPIO states then Wokwi emulator might be a better option than QEMU at this point.

lcgamboa, 2024-12-06:
Hi @mtribiere, I use something similar to your request in the [PICSimLab](https://github.com/lcgamboa/picsimlab) open-source simulator. I use a modified version of [Qemu](https://github.com/lcgamboa/qemu/tree/picsimlab-esp32) that implements GPIO support. The modified version of Qemu is compiled as a library, and the interface is done through callbacks. Take a look at these two functions:  [void qemu_picsimlab_set_pin(int pin, int value)](https://github.com/lcgamboa/qemu/blob/98f1cbda8c214a151a4687eb21dec9daf7f1d0f1/hw/xtensa/esp32_picsimlab.c#L234) that's written on  GPIO Input and [static void pout_irq_handler(void *opaque, int n, int level)](https://github.com/lcgamboa/qemu/blob/98f1cbda8c214a151a4687eb21dec9daf7f1d0f1/hw/xtensa/esp32_picsimlab.c#L260C1-L261C49) callback thas called on ESP32 write on GPIO output. You can link the Qemu library to your code or modify Qemu by adding a socket to communicate with your simulator (I've used this approach, but it makes the simulation much slower).

mtribiere, 2024-12-15:
Hi @igrr, thank you for the sugestion. I did take a look at Wokwi but I'm looking to do everything in a local evironnment, so a web-based tool probably won't cut it.

mtribiere, 2024-12-15:
Hi @lcgamboa thank you for your suggestion. I like the idea of modifying qemu to redirect the gpio callbacks.
If I understand correctly, you trigger an interrupt that will be handled by the `pout_irq_handler` vector is that correct? In that case why not directly overriding the `void qemu_picsimlab_set_pin` function and emitting to the socket from there?

lcgamboa, 2024-12-15:
Hi @mtribiere, the `pout_irq_handler` is called by esp32 to write to an output pin, and `qemu_picsimlab_set_pin` is called by the application to set one esp32 input pin. Take a look at the [old support using sockets](https://github.com/lcgamboa/qemu/blob/picsimlab-stm32/hw/arm/stm32_f103c8_picsimlab.c) (this code is for stm32, but can be modified to be used with esp32).  As I said before the socket version is very slow compared to the direct lib use, but if your use case is only to implement a push-button it is sufficient.