espressif/qemu #110: i2c emulation bug

wodz, 2024-12-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Successful communication with virtual i2c device (tmp105).

### Actual behavior (suspected bug)

WDT reset

### Error logs or terminal output

```shell
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15540
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3904
entry 0x40080640
I (703) boot: ESP-IDF v5.2.1-507-gf8319f08cc 2nd stage bootloader
I (704) boot: compile time Dec 16 2024 09:18:00
I (704) boot: Multicore bootloader
I (722) boot: chip revision: v0.0
I (725) boot.esp32: SPI Speed      : 40MHz
I (725) boot.esp32: SPI Mode       : DIO
I (725) boot.esp32: SPI Flash Size : 4MB
I (734) boot: Enabling RNG early entropy source...
I (744) boot: Partition Table:
I (745) boot: ## Label            Usage          Type ST Offset   Length
I (745) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (745) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (746) boot:  2 factory          factory app      00 00 00010000 00100000
I (753) boot: End of partition table
I (759) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b18ch ( 45452) map
I (783) esp_image: segment 1: paddr=0001b1b4 vaddr=3ffb0000 size=02254h (  8788) load
I (803) esp_image: segment 2: paddr=0001d410 vaddr=40080000 size=02c08h ( 11272) load
I (835) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1703ch ( 94268) map
I (867) esp_image: segment 4: paddr=00037064 vaddr=40082c08 size=0a140h ( 41280) load
I (890) boot: Loaded app from partition at offset 0x10000
I (890) boot: Disabling RNG early entropy source...
I (901) cpu_start: Multicore app
I (1506) cpu_start: Pro cpu start user code
I (1507) cpu_start: cpu freq: 160000000 Hz
I (1507) cpu_start: Application information:
I (1507) cpu_start: Project name:     qemu-firmware
I (1507) cpu_start: App version:      1
I (1508) cpu_start: Compile time:     Dec 16 2024 09:17:53
I (1508) cpu_start: ELF file SHA256:  4bfd61c5f...
I (1508) cpu_start: ESP-IDF:          v5.2.1-507-gf8319f08cc
I (1508) cpu_start: Min chip rev:     v0.0
I (1509) cpu_start: Max chip rev:     v3.99 
I (1509) cpu_start: Chip rev:         v0.0
I (1510) heap_init: Initializing. RAM available for dynamic allocation:
I (1511) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[... 132 lines cut ...]
`

### Project release version

8.2.0-20240122

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Shell

Bash

### Additional context

As can be seen from console logs i2c_master_probe() succeeds but reading from i2c device does not.

igrr, 2024-12-23:
Hi @wodz, thanks for the issue report. I am not very familiar with the I2C peripheral, QEMU emulation was contributed in https://github.com/espressif/qemu/pull/11. Will try to investigate this in the first weeks of January.

One idea, does disabling the watchdogs work in QEMU around the issue?

wodz, 2024-12-24:
> One idea, does disabling the watchdogs work in QEMU around the issue?
No. while(i2c_ll_is_bus_busy(hal->dev)){} is busy loop inside ISR handler in IDF. There is no timeout so firmware waits there indefinitely.

wodz, 2024-12-24:
> Hi @wodz, thanks for the issue report. I am not very familiar with the I2C peripheral, QEMU emulation was contributed in #11. Will try to investigate this in the first weeks of January.

PR says it was tested **writing** to emulated display controller. Write works correctly. The reading part is failing to clear status flags.

Throows, 2024-12-25:
Hello :wave: ,

imo its a ESP-IDF issue. I tried to run the program with the latest version of qemu and ESP-IDF v5.2.1 and i was able to reproduce the bug. 
Then I comiled the latest commit of espressif/qemu on ubuntu 20.10 and tried to reproduce the bug with ESP-IDF v5.3.2 and it seemed to work. :1st_place_medal: 

```
I (647) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (648) boot: compile time Dec 25 2024 12:32:24
I (648) boot: Multicore bootloader
I (655) boot: chip revision: v0.0
I (657) boot.esp32: SPI Speed      : 40MHz
I (657) boot.esp32: SPI Mode       : DIO
I (657) boot.esp32: SPI Flash Size : 2MB
I (660) boot: Enabling RNG early entropy source...
I (665) boot: Partition Table:
I (665) boot: ## Label            Usage          Type ST Offset   Length
I (666) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (666) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (666) boot:  2 factory          factory app      00 00 00010000 00100000
I (669) boot: End of partition table
I (671) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b0e4h ( 45284) map
I (679) esp_image: segment 1: paddr=0001b10c vaddr=3ffb0000 size=02364h (  9060) load
I (685) esp_image: segment 2: paddr=0001d478 vaddr=40080000 size=02ba0h ( 11168) load
I (694) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=171e4h ( 94692) map
I (705) esp_image: segment 4: paddr=0003720c vaddr=40082ba0 size=0a48ch ( 42124) load
I (712) boot: Loaded app from partition at offset 0x10000
I (712) boot: Disabling RNG early entropy source...
I (723) cpu_start: Multicore app
I (1150) cpu_start: Pro cpu start user code
I (1150) cpu_start: cpu freq: 160000000 Hz
I (1151) app_init: Application information:
I (1151) app_init: Project name:     i2c_temp
I (1151) app_init: App version:      1
I (1151) app_init: Compile time:     Dec 25 2024 12:32:22
I (1151) app_init: ELF file SHA256:  4330afa83...
I (1151) app_init: ESP-IDF:          v5.3.2
I (1151) efuse_init: Min chip rev:     v0.0
I (1151) efuse_init: Max chip rev:     v3.99 
I (1152) efuse_init: Chip rev:         v0.0
I (1152) heap_init: Initializing. RAM available for dynamic allocation:
I (1153) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1153) heap_init: At 3FFB2C48 len 0002D3B8 (180 KiB): DRAM
I (1153) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1153) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1153) heap_init: At 4008D02C len 00012FD4 (75 KiB): IRAM
I (1169) spi_flash: detected chip: gd
I (1172) spi_flash: flash io: dio
W (1174) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1179) main_task: Started on CPU0
I (1189) main_task: Calling app_main()
I (1189) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (1189) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (1189) I2C: I2C device found at address 0x48
I (1189) I2C: Temperature raw data: 6400
I (2189) I2C: Temperature raw data: 6400
I (3189) I2C: Temperature raw data: 6400
I (4189) I2C: Temperature raw data: 6400
I (5189) I2C: Temperature raw data: 6400
I (6189) I2C: Temperature raw data: 6400
I (7189) I2C: Temperature raw data: 6400
I (8189) I2C: Temperature raw data: 6400
I (9189) I2C: Temperature raw data: 6400
```

However I am wondering, are we able to control the data of the simulated sensor ?

Best regards

wodz, 2024-12-27:
Yes you can control data sent by the sensor. First you need to add option -qmp unix:/tmp/qmp.sock,server,nowait to the qemu args. Then you have to search for emulated i2c device in qemu object hierarchy using scripts/qmp/qom-list -s /tmp/qmp.sock (I grep for 'temperature' property). Then you can use qom-set -s /tmp/qmp.sock <object>.<property> <value> to set value returned by the sensor. However there is known bug which allows setting this way only string properties so you need to hack qom-set to send int instead of string. Can't find the patch right now but it is pretty trivial.

Regarding IDF vs qemu issue. This is qemu issue revealed by IDF. Reading path of i2c emulation incorrectly returns busy status of the bus BUT indefinitely waiting for some flag in ISR is definitely bad practice.