espressif/qemu #114: Dependency libgcrypt not found (installed v1.11)

safocl, 2025-01-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

My OS archlinux, `libgcrypt 1.11.0-2` and `lib32-libgcrypt 1.11.0-1` (`extra/libgcrypt15 1.5.6-6` and `multilib/lib32-libgcrypt15 1.5.6-8`) are installed.
Installation of `aur/qemu-esp-xtensa-git` package returns:
```console
libgcrypt-config binary missing from cross or native file, or env var undefined.
Trying a default libgcrypt-config fallback at libgcrypt-config
libgcrypt-config found: NO need ['>=1.8']
Run-time dependency libgcrypt found: NO (tried config-tool)

../meson.build:1669:13: ERROR: Dependency "libgcrypt" not found, tried config-tool
```
[meson-log.txt](https://github.com/user-attachments/files/18405811/meson-log.txt)

safocl, 2025-01-14:
Tweak from https://aur.archlinux.org/packages/qemu-esp-xtensa-git#comment-990240 helps to me...

igrr, 2025-01-18:
@lucasssvaz I see you have commented on the linked page, do you know if this is something that has to be fixed in our fork or in archlinux packages?

lucasssvaz, 2025-01-18:
@igrr I created this package long time ago when I was running Arch as my daily driver. I will set up a VM to check what needs to be done.

lucasssvaz, 2025-01-20:
Submitted an internal patch to QEMU to move to `pkg-config`

igrr, 2025-01-20:
@lucasssvaz There is a related upstream commit https://github.com/qemu/qemu/commit/581b4cd5f16d618787bd1e292b851c62c2626da0, could you please check if that solves the problem as well as your patch does? If yes, this will be included in the upcoming rebase on top of upstream v9.2.

igrr, 2025-02-28:
Release [esp-develop-9.2.2-20250228)(https://github.com/espressif/qemu/releases/tag/esp-develop-9.2.2-20250228) includes the above mentioned upstream fix.

closed 2025-02-28 by igrr