espressif/qemu #115: Unable to run RGB Panel example

Joebeazelman, 2025-01-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

RGB Panel should run under QEMU  when following instructions:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/tools/qemu.html

Consistent and accurate error messages and documentation. 

### Actual behavior (suspected bug)

The documentation for running [RGB example](https://github.com/espressif/esp-idf/tree/0f0068ff/examples/peripherals/lcd/rgb_panel) states the RGB example runs on ESP32-S3. Presumably, it should also work under QEMU with graphics. 

I can build the RGB Panel example without issue in VSCode, but when I try to run it under QEMU from VSCode, I get the following error message:

`esp32s3 is not supported by Espressif QEMU. Only esp32 or esp32c3 targets are supported.`

This message is contradicted by the [IDF QEMU Emulator documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/tools/qemu.html). Oddly enough, the documentation states you should install several pre-built libraries for, including **qemu-riscv32**. Is it necessary for ESP32-s3, or is this generalized across SOCs?

When I try it from the command-line using **idf.py build** and  **idf.py qemu -g**, it runs but I get the following message:

```
I (2286) boot: Loaded app from partition at offset 0x10000
I (2286) boot: Disabling RNG early entropy source...
E (2290) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E cpu_start: Failed to init external RAM!

```
Clearly, VSCode and idf.py don't report the same error message. It would be nice if the example documentation included not only the supported SOCs, but a support matrix of QEMU and SOC combination. 

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

See actual behavior.

### Project release version

5.4

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

Sequoia 

### Shell

ZSH

igrr, 2025-01-18:
Hi @Joebeazelman,

I think you have encountered several issues:

> The documentation for running [RGB example](https://github.com/espressif/esp-idf/tree/0f0068ff/examples/peripherals/lcd/rgb_panel) states the RGB example runs on ESP32-S3. Presumably, it should also work under QEMU with graphics.

This is not clearly stated in the example documentation, but this example will not run under QEMU. This example uses the `esp_lcd_panel_rgb` driver, which doesn't support QEMU (or, QEMU doesn't support emulation of the LCD_CAM peripheral in the ESP32-S3`.

[QEMU documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/tools/qemu.html#graphics-support) in ESP-IDF states that:

> QEMU supports a virtual framebuffer device. This device doesn't exist in the real ESP32-S3 hardware, but it can be used to test graphics applications in QEMU.

and

> To use the virtual framebuffer device in your application, you can add the [espressif/esp_lcd_qemu_rgb](https://components.espressif.com/components/espressif/esp_lcd_qemu_rgb) component to your project. This component provides an esp_lcd compatible driver for the virtual framebuffer device.

Therefore the example which can work in QEMU is this one: https://components.espressif.com/components/espressif/esp_lcd_qemu_rgb/versions/1.0.2/examples/lcd_qemu_rgb_panel?language=en.

Please note that if you are using IDF v5.4 and QEMU release esp_develop_9.0.0_20240606 (supplied with ESP-IDF), then this example still won't work on ESP32-S3, since support for virtual LCD framebuffer has only been added in QEMU in commit https://github.com/espressif/qemu/commit/6f41181223a0c8c01571c908045aa5b6ab8f2e89, which hasn't been released yet. As is, the example will work on ESP32 and ESP32-C3.

For this part, we will:

- make the next release of QEMU, including virtual LCD framebuffer support for ESP32-S3 (https://github.com/espressif/qemu/commit/6f41181223a0c8c01571c908045aa5b6ab8f2e89)
- update examples/peripherals/lcd/rgb_panel example readme file with a note that this example doesn't work in QEMU

>  but when I try to run it under QEMU from VSCode, I get the following error message:
> esp32s3 is not supported by Espressif QEMU. Only esp32 or esp32c3 targets are supported.

I think this issue is that in VS Code extension, the list of targets supported by QEMU has been [hard-coded](https://github.com/espressif/vscode-esp-idf-extension/blob/7db37828ebdfe83c49a2bbed6c1af076ef33f755/src/qemu/qemuManager.ts#L150-L158) and hasn't been updated when support for ESP32-S3 in QEMU was added. I'll open an issue in VS Code extension repository to get that fixed.

igrr, 2025-02-28:
QEMU release [esp-develop-9.2.2-20250228](https://github.com/espressif/qemu/releases/tag/esp-develop-9.2.2-20250228) includes support for the virtual RGB framebuffer on ESP32-S3 via [espressif/esp_lcd_qemu_rgb](https://components.espressif.com/components/espressif/esp_lcd_qemu_rgb) component. IDF still needs an update to install this latest release.

Joebeazelman, 2025-03-06:
Thanks. Unfortunately, it's still unclear how you install the latest version of the tool. The **idf_tools.py** doesn't install the tool if you supply a version number such as **esp-develop@9.2.2**.  It also won't install it from an archive downloaded on my local machine. I also noticed it doesn't have any facilities to uninstall a specific tool. 

I also took a look at the **tools.json** file hoping I could change the downloaded version of **esp-develop**, but it's a bit precarious with hardcoded URL version strings and encryption strings. Ironically, in Espressif's monumental effort to automate and consolidate its tools and SDK setup, it has only made the process more difficult. I could unarchive release directly into the tools folder, but it wouldn't perform any special installation steps, which might result in a broken build. 

Personally, I think IDF helps Espressif developers, but hinders end-users by requiring them to reverse engineer a highly complex interconnected web of scripts, tools and SDKs to effectively work with it. It chews up development time by adding unnecessary complexity.

igrr, 2025-03-07:
Hi @Joebeazelman, it's not yet expected that it would be possible to install release v9.2.2 via idf_tools.py — as you correctly noted, an update of tools.json is required for that. We are still validating an upgrade to QEMU 9.2.2 from IDF side and seeing some regressions. I will let you know once we resolve them and the release is integrated into IDF. I'll also reopen this issue until that is done.

Joebeazelman, 2025-03-07:
Thanks. I've managed to make some temporary modifications. Just so you know, the PSRAM isn't configured by default resulting in the following crash:

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x4 (SPI_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (3888) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (3888) boot: compile time Mar  7 2025 12:06:17
I (3888) boot: Multicore bootloader
I (3890) boot: chip revision: v0.0
I (3890) boot: efuse block revision: v0.0
I (3890) boot.esp32s3: Boot SPI Speed : 80MHz
I (3890) boot.esp32s3: SPI Mode       : SLOW READ
I (3890) boot.esp32s3: SPI Flash Size : 2MB
I (3892) boot: Enabling RNG early entropy source...
I (3894) boot: Partition Table:
I (3894) boot: ## Label            Usage          Type ST Offset   Length
I (3894) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (3894) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (3895) boot:  2 factory          factory app      00 00 00010000 00100000
I (3895) boot: End of partition table
I (3895) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=10a30h ( 68144) map
I (3900) esp_image: segment 1: paddr=00020a58 vaddr=3fc94700 size=032d0h ( 13008) load
I (3901) esp_image: segment 2: paddr=00023d30 vaddr=40374000 size=0c2e8h ( 49896) load
I (3906) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=4e0d0h (319696) map
I (3928) esp_image: segment 4: paddr=0007e0f8 vaddr=403802e8 size=04360h ( 17248) load
I (3929) esp_image: segment 5: paddr=00082460 vaddr=600fe100 size=0001ch (    28) load
I (3930) boot: Loaded app from partition at offset 0x10000
I (3930) boot: Disabling RNG early entropy source...
E (3935) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E cpu_start: Failed to init external RAM!

abort() was called at PC 0x40375635 on core 0
--- 0x40375635: call_start_cpu0 at /Users/Joe/esp/v5.4/esp-idf/components/esp_system/port/cpu_start.c:589
```