espressif/qemu #121: Build Failed

Chandler-Kluser, 2025-02-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

A Successful Build

### Actual behavior (suspected bug)

Using my main `gcc` from my archlinux distribution:

```
$ gcc --version
gcc (GCC) 14.2.1 20250207
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

build fails when compiling `block_nfs.c`.

### Error logs or terminal output

```shell
[1154/2566] Compiling C object libblock.fa.p/block_nfs.c.o
FAILED: libblock.fa.p/block_nfs.c.o 
cc -m64 -mcx16 -Ilibblock.fa.p -I. -I.. -Iqapi -Itrace -Iui -Iui/shader -Iblock -I/usr/include/p11-kit-1 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-6 -fdiagnostics-color=auto -Wall -Winvalid-pch -Werror -std=gnu11 -O0 -g -fsanitize=undefined -fsanitize=address -fstack-protector-strong -Wempty-body -Wendif-labels -Wexpansion-to-defined -Wformat-security -Wformat-y2k -Wignored-qualifiers -Wimplicit-fallthrough=2 -Winit-self -Wmissing-format-attribute -Wmissing-prototypes -Wnested-externs -Wold-style-declaration -Wold-style-definition -Wredundant-decls -Wshadow=local -Wstrict-prototypes -Wtype-limits -Wundef -Wvla -Wwrite-strings -Wno-missing-include-dirs -Wno-psabi -Wno-shift-negative-value -isystem /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu/linux-headers -isystem linux-headers -iquote . -iquote /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu -iquote /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu/include -iquote /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu/host/include/x86_64 -iquote /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu/host/include/generic -iquote /home/chandler/SSD2/repositories/xtensa-qemu-test/src/qemu/tcg/i386 -pthread -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -fno-strict-aliasing -fno-common -fwrapv -ftrivial-auto-var-init=zero -fzero-call-used-regs=used-gpr -fPIE -MD -MQ libblock.fa.p/block_nfs.c.o -MF libblock.fa.p/block_nfs.c.o.d -o libblock.fa.p/block_nfs.c.o -c ../block/nfs.c
../block/nfs.c: In function ‘nfs_co_preadv’:
../block/nfs.c:277:29: error: passing argument 3 of ‘nfs_pread_async’ makes pointer from integer without a cast [-Wint-conversion]
  277 |                             offset, bytes, nfs_co_generic_cb, &task) != 0) {
      |                             ^~~~~~
      |                             |
      |                             int64_t {aka long int}
In file included from ../block/nfs.c:49:
/usr/include/nfsc/libnfs.h:723:34: note: expected ‘void *’ but argument is of type ‘int64_t’ {aka ‘long int’}
  723 |                            void *buf, size_t count, uint64_t offset,
      |                            ~~~~~~^~~
../block/nfs.c:277:44: error: passing argument 5 of ‘nfs_pread_async’ makes integer from pointer without a cast [-Wint-conversion]
  277 |                             offset, bytes, nfs_co_generic_cb, &task) != 0) {
      |                                            ^~~~~~~~~~~~~~~~~
      |                                            |
      |                                            void (*)(int,  struct nfs_context *, void *, void *)
/usr/include/nfsc/libnfs.h:723:62: note: expected ‘uint64_t’ {aka ‘long unsigned int’} but argument is of type ‘void (*)(int,  struct nfs_context *, void *, void *)’
  723 |                            void *buf, size_t count, uint64_t offset,
      |                                                     ~~~~~~~~~^~~~~~
../block/nfs.c:277:63: error: passing argument 6 of ‘nfs_pread_async’ from incompatible pointer type [-Wincompatible-pointer-types]
  277 |                             offset, bytes, nfs_co_generic_cb, &task) != 0) {
      |                                                               ^~~~~
      |                                                               |
      |                                                               NFSRPC *
/usr/include/nfsc/libnfs.h:724:35: note: expected ‘nfs_cb’ {aka ‘void (*)(int,  struct nfs_context *, void *, void *)’} but argument is of type ‘NFSRPC *’
  724 |                            nfs_cb cb, void *private_data);
      |                            ~~~~~~~^~
../block/nfs.c:276:13: error: too few arguments to function ‘nfs_pread_async’
  276 |         if (nfs_pread_async(client->context, client->fh,
      |             ^~~~~~~~~~~~~~~
/usr/include/nfsc/libnfs.h:722:12: note: declared here
  722 | EXTERN int nfs_pread_async(struct nfs_context *nfs, struct nfsfh *nfsfh,
      |            ^~~~~~~~~~~~~~~
../block/nfs.c: In function ‘nfs_co_pwritev’:
../block/nfs.c:323:30: error: passing argument 3 of ‘nfs_pwrite_async’ makes pointer from integer without a cast [-Wint-conversion]
  323 |                              offset, bytes, buf,
      |                              ^~~~~~
      |                              |
      |                              int64_t {aka long int}
/usr/include/nfsc/libnfs.h:785:41: note: expected ‘const void *’ but argument is of type ‘int64_t’ {aka ‘long int’}
  785 |                             const void *buf, size_t count, uint64_t offset,
      |                             ~~~~~~~~~~~~^~~
../block/nfs.c:323:45: error: passing argument 5 of ‘nfs_pwrite_async’ makes integer from pointer without a cast [-Wint-conversion]
  323 |                              offset, bytes, buf,
      |                                             ^~~
      |                                             |
      |                                             char *
/usr/include/nfsc/libnfs.h:785:69: note: expected ‘uint64_t’ {aka ‘long unsigned int’} but argument is of type ‘char *’
  785 |                             const void *buf, size_t count, uint64_t offset,
      |                                                            ~~~~~~~~~^~~~~~
[1171/2566] Compiling C object libblockdev.fa.p/nbd_server.c.o
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

1. git clone repository
2. change directory to repository root
3. run `sed -i "s/config-tool/pkg-config/" meson.build`
4. run `./configure --target-list=xtensa-softmmu --enable-gcrypt --enable-debug --enable-sanitizers --disable-strip --disable-user --disable-capstone --disable-vnc --disable-sdl --disable-gtk`
5. build configured recipe generator for ninja

### Project release version

e84c237ddd82efa958c01fc434665a9626ee29bb, on esp-develop branch

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

archlinux

### Shell

ZSH

### Additional context

AUR package maintenance could improve, [this user reported](https://aur.archlinux.org/packages/qemu-esp-xtensa-git#comment-990240) a workaround which is not working so far, yet.

By the way, this is not fixed by AUR maintainer since March,2023...

Now I am building from source with the issue presented.

igrr, 2025-02-28:
This should now be "fixed" by limiting the supported libnfs versions to <6.0.0 (https://github.com/espressif/qemu/commit/42490ac98088a744113589aa75400d9db714776d). This commit is now in `esp-develop` branch of espressif/qemu, following the rebase on top of upstream v9.2.2 version.

Here is the related upstream mailing list [discussion](https://lists.nongnu.org/archive/html/qemu-block/2024-12/msg00151.html) and the [patch](https://lists.nongnu.org/archive/html/qemu-devel/2024-12/msg04282.html).

closed 2025-02-28 by igrr

Chandler-Kluser, 2025-03-01:
> This should now be "fixed" by limiting the supported libnfs versions to <6.0.0 ([42490ac](https://github.com/espressif/qemu/commit/42490ac98088a744113589aa75400d9db714776d)). This commit is now in `esp-develop` branch of espressif/qemu, following the rebase on top of upstream v9.2.2 version.
> 
> Here is the related upstream mailing list [discussion](https://lists.nongnu.org/archive/html/qemu-block/2024-12/msg00151.html) and the [patch](https://lists.nongnu.org/archive/html/qemu-devel/2024-12/msg04282.html).

Thanks a lot for your help and good information, I will check that out!!

Chandler-Kluser, 2025-03-03:
I was able to fully build in a dockerized debian environment, thank you!