espressif/qemu #126: ESP32-S3 crashes on usb_host_install

OOHehir, 2025-04-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hello, I'm using the USB mass storage example in IDF v5.4 but I've found that the ESP32-S3 QEMU panics at this [line](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/msc/main/msc_example_main.c#L425).

I assume it's because the USB is not supported on the QEMU but is there a method at run or compile time for the code to detect that it's running on a QEMU & hence exclude this code?

Thanks,

### Actual behavior (suspected bug)

`I (18273) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (18273) boot: compile time Apr 23 2025 16:51:22
I (18273) boot: Multicore bootloader
I (18275) boot: chip revision: v0.3
I (18275) boot: efuse block revision: v0.0
I (18275) boot.esp32s3: Boot SPI Speed : 80MHz
I (18275) boot.esp32s3: SPI Mode       : DIO
I (18275) boot.esp32s3: SPI Flash Size : 2MB
I (18276) boot: Enabling RNG early entropy source...
I (18277) boot: Partition Table:
I (18278) boot: ## Label            Usage          Type ST Offset   Length
I (18278) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (18278) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (18278) boot:  2 factory          factory app      00 00 00010000 00100000
I (18278) boot: End of partition table
I (18279) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0f4b8h ( 62648) map
I (18284) esp_image: segment 1: paddr=0001f4e0 vaddr=3fc92a00 size=00b38h (  2872) load
I (18285) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=35000h (217088) map
I (18301) esp_image: segment 3: paddr=00055028 vaddr=3fc93538 size=01fb8h (  8120) load
I (18302) esp_image: segment 4: paddr=00056fe8 vaddr=40374000 size=0e954h ( 59732) load
I (18314) esp_image: segment 5: paddr=00065944 vaddr=600fe100 size=0001ch (    28) load
I (18315) boot: Loaded app from partition at offset 0x10000
I (18315) boot: Disabling RNG early entropy source...
I (18318) cpu_start: Multicore app
I (18327) cpu_start: Pro cpu start user code
I (18327) cpu_start: cpu freq: 160000000 Hz
I (18327) app_init: Application information:
I (18327) app_init: Project name:     usb-msc
I (18327) app_init: App version:      1
I (18327) app_init: Compile time:     Apr 23 2025 16:51:17
I (18327) app_init: ELF file SHA256:  d480bebd2...
I (18327) app_init: ESP-IDF:          v5.4-dirty
I (18327) efuse_init: Min chip rev:     v0.0
I (18327) efuse_init: Max chip rev:     v0.99 
I (18327) efuse_init: Chip rev:         v0.3
I (18327) heap_init: Initializing. RAM available for dynamic allocation:
I (18328) heap_init: At 3FC95E28 len 000538E8 (334 KiB): RAM
I (18328) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (18328) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (18328) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (18329) spi_flash: detected chip: winbond
I (18330) spi_flash: flash io: dio
I (18330) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (18330) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (18331) main_task: Started on CPU0
I (18331) main_task: Calling app_main()

assert failed: 0x40380905 <cached disabled>:134 (<cached disabled>)
--- 0x40380905: __assert_func at /home/owen/esp/v5.4/esp-idf/components/newlib/assert.c:58

Backtrace: 0x40375b55:0x3fc9a690 0x4037a1f1:0x3fc9a6b0 0x40380999:0x3fc9a6d0 0x4201d8c2:0x3fc9a7f0 0x42011d99:0x3fc9a810 0x42013e82:0x3fc9a830 0x4200d815:0x3fc9a880 0x4200a629:0x3fc9a900 0x4037abd1:0x3fc9a950
--- 0x40375b55: panic_abort at /home/owen/esp/v5.4/esp-idf/components/esp_system/panic.c:454
0x4037a1f1: esp_system_abort at /home/owen/esp/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40380999: __assert_func at /home/owen/esp/v5.4/esp-idf/components/newlib/assert.c:80
0x4201d8c2: usb_dwc_hal_init at /home/owen/esp/v5.4/esp-idf/components/hal/usb_dwc_hal.c:134 (discriminator 1)
0x42011d99: hcd_port_init at /home/owen/esp/v5.4/esp-idf/components/usb/hcd_dwc.c:1279
0x42013e82: hub_install at /home/owen/esp/v5.4/esp-idf/components/usb/hub.c:580
0x4200d815: usb_host_install at /home/owen/esp/v5.4/esp-idf/components/usb/usb_host.c:532
0x4200a629: usb_task at /home/projects/msc/main/msc_example_main.c:218
0x4037abd1: vPortTaskWrapper at /home/owen/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: d480bebd2

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x4 (SPI_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
`

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

source $IDF_PATH/export.sh 
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Activating ESP-IDF 5.4
* Checking python version ... 3.10.12
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... bash
* Detecting outdated tools in system ... OK - no outdated tools found
* Shell completion ... Autocompletion code generated

### Project release version

v5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 

### Shell

Bash

igrr, 2025-04-29:
There is no readily available API in IDF for run-time detection of QEMU.

One option is to set some bit in eFuse (before launching QEMU) and then read this bit from the application. Another option might be to read some memory-mapped register which has different value between the real chip and the emulator (e.g. UART_DATE_REG).

As for compile-time selection, you can define a Kconfig option (e.g. BUILD_FOR_QEMU) in your application and set it from menuconfig, then check it at compile time.

OOHehir, 2025-04-29:
@igrr Many thanks for the quick response! I'll try those suggestions.

closed 2025-04-29 by OOHehir

igrr, 2025-04-29:
Forgot to add, here are a couple of examples which may help:

https://github.com/espressif/idf-extra-components/blob/ebd450a45c6485e5ef73ebfcd05761c2d1817def/esp_lcd_qemu_rgb/src/esp_lcd_qemu_rgb.c#L53-L56

https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/components/esp_eth/src/openeth/esp_eth_mac_openeth.c#L298-L303

OOHehir, 2025-04-29:
Great, thanks!