espressif/qemu #132: ESP32-S3: Flash Encryption is broken (IDFGH-15811)

dmitrij9992905, 2025-07-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2, master

### Espressif SoC revision.

ESP32-S3 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

QEMU, ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

If flash encryption is enabled in sdkconfig and no key is burned initially, then generate encryption key, encrypt necessary partitions, then boot normally with flash encrypted; (tested with both AES-XTS-128 and AES-XTS-256)
If flash encryption is enabled and key is pre-burned, then encrypt necessary partitions using this key then boot normally (tested only AES-XTS-128)

### What is the actual behavior?

In both cases, partitions are being encrypted, sending restart signal... and it hangs
After rebooting, invalid header messages occures.
On other chips that can be emulated in QEMU (esp32, esp32-c3) it works normally.
Tried both with and without Secure Boot.

### Steps to reproduce.

1. config the project to flash encryption
2. idf.py build
3. idf.py qemu monitor

### Debug Logs.

```plain
sdkconfig:
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="prod_secure_boot_key.pem"
[... 90 lines cut ...]
I (1397) esp_image: segment 5: paddr=0041d328 vaddr=600fe100 size=0001ch (    28) 
I (1398) esp_image: segment 6: paddr=0041d34c vaddr=00000000 size=02c84h ( 11396) 
I (1402) esp_image: Verifying image signature...
I (1404) secure_boot_v2: Verifying with RSA-PSS...
I (1406) secure_boot_v2: Signature verified successfully!
I (1407) flash_encrypt: Encrypting partition 3 at offset 0x270000 (length 0x2c0000)...
I (4369) flash_encrypt: Done encrypting
E (4369) esp_image: image at 0x530000 has invalid magic byte (nothing flashed here?)
I (4370) flash_encrypt: Encrypting partition 5 at offset 0x7f0000 (length 0x10000)...
I (4446) flash_encrypt: Done encrypting
I (4447) efuse: BURN BLOCK0
I (4451) efuse: BURN BLOCK0 - OK (all write block bits are set)
I (4453) flash_encrypt: Flash encryption completed
I (4453) boot: Resetting with flash encryption enabled...
<< Monitor log freezes here, and after repeated 'idf.py -B "sdkconfig.secure" qemu monitor' returns invalid header >>

Other details in the forum: https://esp32.com/viewtopic.php?t=46038
```

### More Information.

Also tried in Windows (CMD and PowerShell)

Logs in Windows
[esp32s3 flash encryption secure boot log.txt](https://github.com/user-attachments/files/21087311/esp32s3.flash.encryption.secure.boot.log.txt)

dmitrij9992905, 2025-07-07:
As for QEMU now it works properly with flash encryption on QEMU 9.2.2
As for real chip, now I have v0.2 revision, and I pre-generated a flash encryption key, burned this one into chip, flashed a full binary.
Encrypted and started OK onboard, but I cannot encrypt properly on host using esptool tools (like esptool, espsecure), v4.8.1 used.

Logs are in attachments:

[esp32s3 flash encryption logs device 07072025.txt](https://github.com/user-attachments/files/21101650/esp32s3.flash.encryption.logs.device.07072025.txt)
[esp32s3 flash encryption logs esptool 07072025.txt](https://github.com/user-attachments/files/21101649/esp32s3.flash.encryption.logs.esptool.07072025.txt)

Harshal5, 2025-07-08:
>  using --aes-xts flag helped me with host partitions encryption

Right.

You could check out the host-based (externally) security features enablement documentation [here](https://docs.espressif.com/projects/esp-idf/en/v5.4.2/esp32s3/security/security-features-enablement-workflows.html). It contains all the information related to externally enabled Secure Boot, Flash Encryption and NVS Encryption.

Harshal5, 2025-07-08:
Feel free to open an issue in the ESP-IDF repository if you have any questions regarding the host-based (externally) security features enablement documentation/workflow. In case you have no further doubts regarding the QEMU usage, feel free to close this issue as well!

dmitrij9992905, 2025-07-08:
Figured out how to encrypt correctly.
I'll try to flash pre-encrypted image, then burn efuses regarding to flash encryption, but leave secureboot efuses as them are, to be burnt automatically at startup by bootloader.
For now, closing issue, and if needed, then it will be reopened.

closed 2025-07-08 by dmitrij9992905